google-deepmind/dm_control #255: The GLFW library is not initialized

seolhokim, 2022-02-18:
## Problem

`from dm_control import suite`
warning 
`/home/seolho/anaconda3/envs/main/lib/python3.7/site-packages/glfw/__init__.py:906: GLFWError: (65544) b'X11: The DISPLAY environment variable is missing'`

I ignored and called suite.load
```
env = suite.load("hopper","hop")
env.reset()
```
result
``` 
TimeStep(step_type=<StepType.FIRST: 0>, reward=None, discount=None, observation=OrderedDict([('position',array([-0.01262767, -0.99996228,  0.00868512])), ('velocity', array([-0.00717681, -0.01415571]))]))
```
But It doesn't work

```
from dm_control.suite.wrappers import pixels
env_pixels = pixels.Wrapper(env)
```

result
```
~~~/glfw/__init__.py:906: GLFWError: (65537) b'The GLFW library is not initialized'
  warnings.warn(message, GLFWError)
python: /builds/florianrhiem/pyGLFW/glfw-3.3.5/src/posix_thread.c:64: _glfwPlatformGetTls: Assertion `tls->posix.allocated == 1' failed.
Aborted (core dumped)
```
and python turned off.

I set environment variable
`export DISPLAY=0`
result
`
/home/seolho/anaconda3/envs/main/lib/python3.7/site-packages/glfw/__init__.py:906: GLFWError: (65544) b'X11: Failed to open display 1`
What can I do?

kevinzakka, 2022-02-23:
Have you tried setting `MUJOCO_GL` to either `egl` or `osmesa`?

seolhokim, 2022-02-28:
Thanks 
``` 
import os
os.environ['MUJOCO_GL'] = 'osmesa'
```
works for me.

kevinzakka, 2022-02-28:
@seolhokim I'd recommend you try to get `egl` to work as you get non-trivial speedups in rendering.

Try running `sudo apt install libglfw3 libglew2.1 libglew-dev` and `MUJOCO_GL='egl'` to test it.

seolhokim, 2022-02-28:
@kevinzakka 
Very Thanks! It also works well. I need to reduce computation time, It will be very helpful for me.

closed 2022-02-28 by seolhokim