google-deepmind/dm_control #257: Libmujoco.so.2.1.1: cannot open shared object file: No such file or directory

gemcollector, 2022-02-24:
when I run:
`from dm_control import suite`
the error is:
`OSError: /home/user/.mujoco/mujoco-2.1.1/lib/libmujoco.so.2.1.1: cannot open shared object file: No such file or directory`
I don't know how to fix this bug, thx~!

saran-t, 2022-02-24:
Have you downloaded and unpacked MuJoCo into `/home/user/.mujoco`? This is described in our README.

gemcollector, 2022-02-24:
> 

yeah, I did it, my environment variables are:
`
export MUJOCO_PY_MJKEY_PATH=/home/yzc_20/.mujoco/mjkey.txt
export MUJOCO_PY_MUJOCO_PATH=/home/yzc_20/.mujoco/mujoco200/
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/yzc_20/.mujoco/mujoco200/bin/
export MJLIB_PATH=/home/yzc_20/.mujoco/mujoco-2.1.1/lib/libmujoco.so.2.1.1
`

gemcollector, 2022-02-25:
> ```shell
> libglew-dev
> ```

thx for your rely!~ But I still can't run 
`from dm_control import suite`
with the same error. It shows that "No such file or directory", but I truly have this file! By the way , the following command is ok:
`import mujoco-py`

I don't know what's the problem?

gemcollector, 2022-02-25:
I find the problem. It is important to install a correct vesion of mujoco-2.1.1:
`mujoco-2.1.1-linux-x86_64.tar.gz`
I didn't install a x86 version. @kevinzakka thx for your instruction and thx @saran-t for your reply!
I have solved problem!

kevinzakka, 2022-02-25:
@saran-t Slight tangent, but there are so many issues with installing mujoco and dm_control, I wonder if we could make an interface like pytorch where instructions are clearly given based on OS/installation method:

[image]