google-deepmind/dm_control #260: pyMJCF: Overriding visual/global/offheight

ishank-juneja, 2022-03-04:
I wanted to get 512x512 frames out of my mujoco simulation created using a pyMJCF model.

When I run,
```
pixels = physics.render(height=512, width=512, camera_id=0, depth=False)
```

I get a buffer length not enough error (512 > default offheight of 480) so I tried changing the setting through the pyMJCF global attributes but none of the below worked,
```
self._model = mjcf.RootElement(model="regular-polygon")
self._model.visual.global.offheight = 512 (Error: visual has no attribute global)
self._model.visual.global_.offheight = 512 (Error: visual has no attribute global_)
self._model.vis.global_ = 512 (Error: rootElement has no attribute vis)
``` 

Please tell me how I can increase the offheight buffer to 512 without having to resort to loading in an xml backbone

yuvaltassa, 2022-03-04:
Unfortunately I believe this is the only way at the moment. 
```
self._model.visual.__getattr__('global').offheight
```

closed 2022-03-04 by yuvaltassa

vaxenburg, 2024-07-26:
I think this also works:
```python
self._model.visual.get_children('global').offheight
```