google-deepmind/dm_control #261: Mocap body

matansudry, 2022-03-08:
Hi,

I am trying to add a mocap body on a rope, and it seems the mocap is not connected, I tried to use weld and connect.

<mujoco model="rope">
    
    <include file="scene_v2.xml"/>

    <option timestep="0.002" jacobian="dense"/>

    <size nconmax="100" njmax="300" nstack="50000"/>

    <worldbody>
        <body name="B10" pos="0 -0.5 0.00">
            <freejoint/>
            <composite type="rope" count="21 1 1" spacing="0.04" offset="0 0 2">
                <joint kind="main" damping="0.005"/>
                <geom  type="capsule" size=".01 .015" rgba=".8 .2 .1 1"/>
            </composite>
        </body>
        <body name="hand" mocap="true" pos="0 -0.5 0.00">
            <geom type="box" size=".01 .01 .01" rgba="0 .9 0 .5" contype="2"/>
        </body>
    </worldbody>
    <equality>
        <weld body1="hand" body2="B10" solimp="0.9 0.95 0.001" solref="0.02 1"></weld>
    </equality>
</mujoco>

yuvaltassa, 2022-03-08:
This looks like it should work. Could you please attach a zip with everything needed to load the model so we can look? (scene_v2.xml, any textures or meshes etc)

matansudry, 2022-03-08:
please see below.

scene_v2.xml
<mujoco>
    <statistic extent="2" meansize=".05"/>

    <option timestep="0.005" solver="Newton" iterations="30" tolerance="1e-10" jacobian="sparse" cone="pyramidal"/>

    <visual>
        <rgba haze="0.15 0.25 0.35 1"/>
        <quality shadowsize="2048"/>
        <map stiffness="700" shadowscale="0.5" fogstart="10" fogend="15" zfar="40" haze="0.3"/>
    </visual>

    <asset>
        <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="512"/> 
        <texture name="texplane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2" 
            width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>  
        <texture name="texcarpet" type="2d" file="carpet.png"/>  
        <texture name="texsponge" type="2d" file="sponge.png"/>  
        <texture name="texmarble" type="cube" file="marble.png"/>  

        <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
        <material name="matcarpet" texture="texcarpet"/>
        <material name="matsponge" texture="texsponge" specular="0.3"/>
        <material name="matmarble" texture="texmarble" rgba=".7 .7 .7 1"/>
    </asset>

    <worldbody>
        <light directional="true" diffuse=".4 .4 .4" specular="0.1 0.1 0.1" pos="0 0 5.0" dir="0 0 -1" castshadow="false"/>
        <light directional="true" diffuse=".6 .6 .6" specular="0.2 0.2 0.2" pos="0 0 4" dir="0 0 -1"/>
        <geom name="ground" type="plane" size="0 0 1" pos="0 0 0" quat="1 0 0 0" material="matplane" condim="1"/>
    </worldbody>
</mujoco>

[image]
[image]
[image]

matansudry, 2022-03-11:
The connections seem to be working, but I have faced a new problem.
In my scenario, I want to connect the mocap body to each link in the rope and move this body according to some policy.
If I choose mocap N, the other mocaps will interrupt me when I am moving the object because the links they are connected to will push to stay on some place.
I need to find one of two solutions:
1. connect and disconnect the mocap body during the physics is alive. By doing that, I can move it to the right link for each action
2. Disable any influence of N-1 mocaps, and only the chosen one can move my rope.

What is feasible in mujoco?

yuvaltassa, 2022-03-14:
You can enable/disable specific equality constraints by flipping the flags in `mjModel.eq_active` at runtime.

closed 2022-03-14 by yuvaltassa