google-deepmind/dm_control #262: `mjlib.mjr_figure` in `Viewer` object to render 2D plots in realtime

jimzers, 2022-03-08:
Hi, was wondering if there was a way to render a 2D plot in realtime with the viewer. I looked into `mjlib.mjr_figure` and looked at the example for [`simulate.cc`](https://github.com/deepmind/mujoco/blob/main/sample/simulate.cc), but wasn't sure what format to put the image in.

yuvaltassa, 2022-03-08:
You want to do this in Python or C++?

If you run `dm_control/suite/explore.py` you should see the the window has some live 2d plots. (At least I think it does? There are some differences between the internal and external versions, but I think it should. Try some F-keys) You can follow these and see how the rendering commands are implemented.

In simulate you can see the native plots by hitting e.g. F3 to see the timing plots that are beings drawn via C++ commands.

The mjUI interface is not well documented yet, so you'll just have to see how we do it, for now.

closed 2022-03-08 by yuvaltassa

kevinzakka, 2022-03-08:
To add to @yuvaltassa , here's an example of how to modify the `dm_control.viewer` in python. We're going to render a random RGB image in the bottom right position of the screen by sampling a 384x384x3 numpy array at every timestep.

In `viewer.views.py`:
```python
class CustomImageView(renderer.Component):
  def render(self, context, viewport):
    width_adjustment = viewport.width % 4
    pos = types.MJRRECT(
        int(3 * viewport.width / 4) + width_adjustment, 0,
        int(viewport.width / 4), int(viewport.height / 4))
    rand_img = np.random.randint(0, 255, size=(pos.height, pos.width, 3), dtype=np.uint8)
    mjlib.mjr_drawPixels(rand_img, None, pos, context.ptr)
```

In `viewer.application.py`, add:

```python
# in __init__.py
self._custom_image_layout = views.CustomImageView()
# in _perform_deferred_reload
self._renderer.components += self._custom_image_layout
```

The result:

https://user-images.githubusercontent.com/10518920/157345884-bbe0adc9-b454-4b40-a786-e6e248a64374.mp4

Should be a good starting point to render custom images / matplotlib plots :-)

rohit-kumar-j, 2022-05-30:
@jimzers For a rudimentary example see [examples/graphs.py](https://github.com/rohit-kumar-j/mujoco-python-viewer/blob/extended-example-tutorial/examples/graphs.py) in this branch. 
You can add custom **math graphs** along with **sensor data**, etc (Although the code isn't clean ...yet)

**Preview:**
The sinewave is a custom data_stream with `math.sin(x)`
The rest of them are sensor data

https://user-images.githubusercontent.com/37873142/171004578-29375d1e-2856-4ded-9c61-77bf9bacc211.mp4