google-deepmind/dm_control #268: Maximizing `viewer` window raises RuntimeError

kevinzakka, 2022-03-15:
As the title suggests, trying to maximize the window size of the `dm_control.viewer` application now crashes with a `RuntimeError`. I've added the traceback below!

<details>
<summary>Toggle traceback</summary>

  ```python
Traceback (most recent call last):
  File "shadow_hand/manipulation/explore.py", line 62, in <module>
    app.run(main)
  File "/Users/kevin/miniconda3/envs/muj/lib/python3.8/site-packages/absl/app.py", line 312, in run
    _run_main(main, args)
  File "/Users/kevin/miniconda3/envs/muj/lib/python3.8/site-packages/absl/app.py", line 258, in _run_main
    sys.exit(main(argv))
  File "shadow_hand/manipulation/explore.py", line 58, in main
    viewer.launch(env, policy=random_policy)
  File "/Users/kevin/miniconda3/envs/muj/lib/python3.8/site-packages/dm_control/viewer/__init__.py", line 40, in launch
    app.launch(environment_loader=environment_loader, policy=policy)
  File "/Users/kevin/miniconda3/envs/muj/lib/python3.8/site-packages/dm_control/viewer/application.py", line 327, in launch
    self._window.event_loop(tick_func=tick)
  File "/Users/kevin/miniconda3/envs/muj/lib/python3.8/site-packages/dm_control/viewer/gui/glfw_gui.py", line 286, in event_loop
    pixels = tick_func()
  File "/Users/kevin/miniconda3/envs/muj/lib/python3.8/site-packages/dm_control/viewer/application.py", line 325, in tick
    self._tick()
  File "/Users/kevin/miniconda3/envs/muj/lib/python3.8/site-packages/dm_control/viewer/application.py", line 271, in _tick
    self._viewer.render()
  File "/Users/kevin/miniconda3/envs/muj/lib/python3.8/site-packages/dm_control/viewer/viewer.py", line 238, in render
    self._renderer.render(self._viewport, scene)
  File "/Users/kevin/miniconda3/envs/muj/lib/python3.8/site-packages/dm_control/viewer/renderer.py", line 151, in render
    self._mujoco_context = wrapper.MjrContext(
  File "/Users/kevin/miniconda3/envs/muj/lib/python3.8/site-packages/dm_control/mujoco/wrapper/core.py", line 605, in __init__
    with gl_context.make_current() as ctx:
  File "/Users/kevin/miniconda3/envs/muj/lib/python3.8/contextlib.py", line 113, in __enter__
    return next(self.gen)
  File "/Users/kevin/miniconda3/envs/muj/lib/python3.8/site-packages/dm_control/_render/base.py", line 121, in make_current
    with self._render_executor.execution_context() as ctx:
  File "/Users/kevin/miniconda3/envs/muj/lib/python3.8/contextlib.py", line 113, in __enter__
    return next(self.gen)
  File "/Users/kevin/miniconda3/envs/muj/lib/python3.8/site-packages/dm_control/_render/executor/render_executor.py", line 89, in execution_context
    self._check_not_terminated()
  File "/Users/kevin/miniconda3/envs/muj/lib/python3.8/site-packages/dm_control/_render/executor/render_executor.py", line 84, in _check_not_terminated
    raise RuntimeError(_ALREADY_TERMINATED)
RuntimeError: This executor has already been terminated.
```

</details>

saran-t, 2022-03-15:
Can you try setting the environment variable `DISABLE_RENDER_THREAD_OFFLOADING=1` ?

kevinzakka, 2022-03-15:
@saran-t Not working, any resize / maximization crashes.

closed 2022-03-23 by saran-t