google-deepmind/dm_control #270: how to properly add mesh assets from files in pymjcf?

stracovskylu, 2022-03-17:
I've been trying to construct mujoco models through pymjcf, and have run into some issues with importing meshes from files. What is the proper way to use pymjcf to add mesh assets?

I've linked a [colab](https://github.com/stracovskylu/misc_mujoco_test/blob/main/pymjcf_mesh_test.ipynb) notebook which has example code for two methods I attempted.

kevinzakka, 2022-03-18:
I get the following error:

`ValueError: Error: number of faces should be between 1 and 200000 in STL file 'Single-part/3DBenchy.stl'; perhaps this is an ASCII file?`

stracovskylu, 2022-03-18:
You are right. I took a look at the stl files I am working with and most are slightly over 200000 faces - and so is the benchy3d model. My mistake. Is there any setting or configuration to increase this maximum allowed value? The meshes I am working with are paired with contact maps, so I am hoping to not have to downsample both the mesh and the contact map.

closed 2022-03-22 by stracovskylu