google-deepmind/dm_control #273: AttributeError when importing viewer

Robokan, 2022-03-19:
Getting an Attribute error about certification question? Any ideas? I can't find any references on the internet about this problem.
using mujoco 2.1.1 and juypter notebooks

I tested mujoco and it works great. dm_control works until I try to import viewer

import dm_control
import mujoco
In [7]:

from dm_control import locomotion
In [9]:

from dm_control import viewer
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Input In [9], in <cell line: 1>()
----> 1 from dm_control import viewer

File ~/miniconda3/envs/rl_course/lib/python3.10/site-packages/dm_control/viewer/__init__.py:22, in <module>
     19 from __future__ import division
     20 from __future__ import print_function
---> 22 from dm_control.viewer import application
     25 def launch(environment_loader, policy=None, title='Explorer', width=1024,
     26            height=768):
     27   """Launches an environment viewer.
     28 
     29   Args:
   (...)
     40       ValueError: When 'environment_loader' argument is set to None.
     41   """

File ~/miniconda3/envs/rl_course/lib/python3.10/site-packages/dm_control/viewer/application.py:25, in <module>
     23 from dm_control import _render
     24 from dm_control.viewer import gui
---> 25 from dm_control.viewer import renderer
     26 from dm_control.viewer import runtime
     27 from dm_control.viewer import user_input

File ~/miniconda3/envs/rl_course/lib/python3.10/site-packages/dm_control/viewer/renderer.py:24, in <module>
     21 import abc
     22 import contextlib
---> 24 from dm_control.mujoco import wrapper
     25 from dm_control.mujoco.wrapper import mjbindings
     26 from dm_control.viewer import util

File ~/miniconda3/envs/rl_course/lib/python3.10/site-packages/dm_control/mujoco/__init__.py:18, in <module>
      1 # Copyright 2017 The dm_control Authors.
      2 #
      3 # Licensed under the Apache License, Version 2.0 (the "License");
   (...)
     13 # limitations under the License.
     14 # ============================================================================
     16 """Mujoco implementations of base classes."""
---> 18 from dm_control.mujoco.engine import action_spec
     20 from dm_control.mujoco.engine import Camera
     21 from dm_control.mujoco.engine import MovableCamera

File ~/miniconda3/envs/rl_course/lib/python3.10/site-packages/dm_control/mujoco/engine.py:45, in <module>
     42 from absl import logging
     44 from dm_control import _render
---> 45 from dm_control.mujoco import index
     46 from dm_control.mujoco import wrapper
     47 from dm_control.mujoco.wrapper import util

File ~/miniconda3/envs/rl_course/lib/python3.10/site-packages/dm_control/mujoco/index.py:92, in <module>
     89 import collections
     90 import weakref
---> 92 from dm_control.mujoco.wrapper import util
     93 from dm_control.mujoco.wrapper.mjbindings import sizes
     94 import numpy as np

File ~/miniconda3/envs/rl_course/lib/python3.10/site-packages/dm_control/mujoco/wrapper/__init__.py:23, in <module>
     19 from __future__ import division
     20 from __future__ import print_function
---> 23 from dm_control.mujoco.wrapper import mjbindings
     25 from dm_control.mujoco.wrapper.core import callback_context
     26 from dm_control.mujoco.wrapper.core import enable_timer

File ~/miniconda3/envs/rl_course/lib/python3.10/site-packages/dm_control/mujoco/wrapper/mjbindings/__init__.py:32, in <module>
     30 # pylint: disable=g-import-not-at-top
     31 try:
---> 32   from dm_control.mujoco.wrapper.mjbindings import functions
     33   from dm_control.mujoco.wrapper.mjbindings.functions import mjlib
     34   logging.info('MuJoCo library version is: %d', mjlib.mj_version())

File ~/miniconda3/envs/rl_course/lib/python3.10/site-packages/dm_control/mujoco/wrapper/mjbindings/functions.py:45, in <module>
     42 mjlib.mj_deactivate.argtypes = None
     43 mjlib.mj_deactivate.restype = None
---> 45 mjlib.mj_certQuestion.__doc__ = """
     46 Server side: generate certificate question.
     47 
     48 Args:
     49   question: util.ndptr(shape=(16,), dtype=np.float64, flags=('C', 'W'))
     50 """
     51 mjlib.mj_certQuestion.argtypes = [
     52     util.ndptr(shape=(16,), dtype=np.float64, flags=('C', 'W')),
     53 ]
     54 mjlib.mj_certQuestion.restype = None

File ~/miniconda3/envs/rl_course/lib/python3.10/ctypes/__init__.py:387, in CDLL.__getattr__(self, name)
    385 if name.startswith('__') and name.endswith('__'):
    386     raise AttributeError(name)
--> 387 func = self.__getitem__(name)
    388 setattr(self, name, func)
    389 return func

File ~/miniconda3/envs/rl_course/lib/python3.10/ctypes/__init__.py:392, in CDLL.__getitem__(self, name_or_ordinal)
    391 def __getitem__(self, name_or_ordinal):
--> 392     func = self._FuncPtr((name_or_ordinal, self))
    393     if not isinstance(name_or_ordinal, int):
    394         func.__name__ = name_or_ordinal

AttributeError: dlsym(0x20a71aae0, mj_certQuestion): symbol not found

saran-t, 2022-03-19:
Please double check that you're using the latest version of `dm_control` (1.0.0).

Robokan, 2022-03-20:
I used the command:

Pip install dm_control

But when I look at the version with pip list it shows:

Version: 0.0.319497192

???

I am using MacOS with M1 processor.

saran-t, 2022-03-20:
I can't comment exactly on what's happening, but perhaps you have an old version installed already. Try `pip install --upgrade`, or uninstall dm_control and reinstall it again.

Robokan, 2022-03-21:
I uninstalled and reinstalled no luck.

so I just typed:

pip index versions dm_control

WARNING: pip index is currently an experimental command. It may be removed/changed in a future release without prior warning.
dm_control (0.0.319497192)
Available versions: 0.0.319497192, 0.0.318066097, 0.0.318037100, 0.0.312466143, 0.0.300771433, 0.0.295778102, 0.0.288483845, 0.0.288398964, 0.0.286955599, 0.0.286587932

It doesn't appear that version 1.0.0 is available at all. Any ideas? I appreciate your help.

Robokan, 2022-03-21:
Ok I fixed everything and it all works. Just had to use the right version of Python and install these:

pkg-config
HDF5

on MacOS these need to be installed before dm_control or it will fail to build.

saran-t, 2022-03-21:
Yeah, HDF5 doesn't ship wheels for M1 Macs. I'll make it an optional dependency for our next release so that it doesn't block people. Thanks.

closed 2022-03-21 by saran-t