google-deepmind/dm_control #275: Unexpected changes to maze arenas since `dm-control==1.0.0`

hartikainen, 2022-03-23:
Hey folks,

I have a few environments that use the `dm_control.locomotion.arenas.mazes` and I'm seeing unexpected changes in these arenas after upgrading to `dm-control==1.0.0`. Specifically, there appears to be some issue with the orientations of the wall geoms. Here's what I see with `dm-control==0.0.425341097`:
[image]

Here's, on the other hand, what I see with `dm-control==1.0.0` using the same code:
[image]

I briefly went through the changes between those two version but nothing obvious stood out. I'm happy to dig into this my self but if anyone happens to know what the problem/solution could be off the top of their head, that might save me some time.

For reproducibility, here's the code I used for this:
```
from absl import app
import numpy as np

from dm_control import viewer
from dm_control.locomotion.examples import basic_cmu_2019

def main(argv):
    del argv

    environment = basic_cmu_2019.cmu_humanoid_maze_forage()

    def random_policy(time_step):
        del time_step

        action_spec = environment.action_spec()
        action = np.random.uniform(
            low=action_spec.minimum,
            high=action_spec.maximum,
            size=action_spec.shape)

        return action

    viewer.launch(
        environment_loader=lambda: environment,
        policy=random_policy)

if __name__ == '__main__':
  app.run(main)
```

saran-t, 2022-03-23:
Hi Kristian! We've noticed this internally as well a couple of days ago. It's a bug in MuJoCo 2.1.2, and we're planning to push a 2.1.3 update (alongside dm_control 1.0.1) out to fix this today.

saran-t, 2022-03-23:
I'm going to transfer this bug across to MuJoCo since it's an upstream issue.