google-deepmind/dm_control #276: viewer not working on mac osx

enajx, 2022-04-01:
Using the latest install for both dm_control and mujoco via pypi, the viewer can not render a window on OSX (intel chip).

Script:

```
from dm_control import suite
from dm_control import viewer
import numpy as np

env = suite.load(domain_name="point_mass", task_name="easy")
action_spec = env.action_spec()

# Define a uniform random policy.
def random_policy(time_step):
  del time_step  # Unused.
  return np.random.uniform(low=action_spec.minimum,
                           high=action_spec.maximum,
                           size=action_spec.shape)

# Launch the viewer application.
viewer.launch(env, policy=random_policy)
```

Error output:

```
template_render_dm.py 51 <module>
"/Users/user/REAL/CMAES_gym/template_render_dm.py", line 51
viewer.launch(env, policy=random_policy)

__init__.py 39 launch
"/Users/user/.virtualenvs/p39/lib/python3.9/site-packages/dm_control/viewer/__init__.py", line 39
app = application.Application(title=title, width=width, height=height)

application.py 187 __init__
"/Users/user/.virtualenvs/p39/lib/python3.9/site-packages/dm_control/viewer/application.py", line 187
self._window = gui.RenderWindow(width, height, title)

__init__.py 34 ErrorRenderWindow
"/Users/user/.virtualenvs/p39/lib/python3.9/site-packages/dm_control/viewer/gui/__init__.py", line 34
raise ImportError(

ImportError:
Cannot create a window because no windowing system could be imported
```

gizemozd, 2022-04-07:
If your macOS version is Big Sur, this issue might be related to the location of OpenGL libraries.

In my case, the following solved my issue. 

Change the following line in ```ctypesloader.py```:
```fullName = util.find_library( name )```
with:
```fullName = "/System/Library/Frameworks/{}.framework/{}".format(name,name) ```

NOTE: ctypesloader.py is under ```~/<virtual-env>/lib/python3.9/site-packages/OpenGL/platform/ctypesloader.py``` or something similar.

[Source](https://stackoverflow.com/a/64021312)

saran-t, 2022-04-07:
Since the original post is from last week, I presume you're still working with `mujoco` version 2.1.3?

If that's the case could you please try upgrading to 2.1.4 to see if the issue persists? (In particular, libGLEW is no longer required).

enajx, 2022-04-07:
I just upgraded to 2.1.4, same issue sadly persists