google-deepmind/dm_control #278: Multi-Objective Policy Optimization in the repo?

MotorCityCobra, 2022-04-01:
[This paper is about an algorithm called Multi-Objective Policy Optimization](https://arxiv.org/abs/2005.07513). It is linked to this repo as if it was implemented here. I would like to see this implemented in Python, but I don't see it. Can I ask where it is exactly?

I have a broader question and I think you at DeepMind are the ones to ask. What is the ideal algorithm for controlling a robot? Smooth control with minimum movement and force applied.  
I'd rather not spend a few months digging deep into something when there is a much better algorithm out there.

nimrod-gileadi, 2022-04-05:
There is an implementation of MO-MPO here: https://github.com/deepmind/acme/tree/master/acme/agents/tf/mompo

The ideal algorithm really depends on your setting, and there is no one answer. That's why people do robotics research.

If you have a perfect model of your robot and environment, and a dense objective function, maybe you want to do optimal control. You'll certainly get smooth control with minimal movement that way.
If you are learning the task in simulation and have a lot of compute, something like PPO or TRPO is good.
If you are looking to be more data efficient, maybe an off policy algorithm like MPO is the right answer.
If you only have offline data, then there are offline RL algorithms.

etc., etc..
MO-MPO fits in a niche where you want to explore a trade off between multiple objectives, e.g run fast, and perform small actions.

closed 2022-08-04 by MotorCityCobra

HONG-ZI, 2022-11-03:
Hello, I have some questions regarding the experiments in the paper “A Distributional View on Multi-Objective Policy Optimization”. 
1. What is the computer configuration used in the experiments? for example how many CPUs and how many memories?
2. How long does each experiment take to train? 

Because I do not have a high-performance workstation, I want to know whether a student can quickly reproduce the results of this paper.

nimrod-gileadi, 2022-11-03:
You could reach out to the main authors (their emails are listed on the paper). They don't follow issues on dm_control's issue tracker.