google-deepmind/dm_control #283: mujoco.FatalError: gladLoadGL error

atonkamanda, 2022-04-06:
Hello, despite the fact that I installed all required dependencies, I get this error again and again. Does someone have any clue about how I could fix it ? Or where it comes from ? 

[image]

Thank you in advance.

atonkamanda, 2022-04-08:
Fixed it I think the new mujoco implementation from deepmind make some dependencies inconsistent,(the pip install mujoco) so you need to install it the old way with paths like here https://github.com/openai/mujoco-py .

saran-t, 2022-04-08:
That doesn't seem possible since our new bindings isn't relying on any path environment variable to look for any library. What environment variable did you set that made things work?

atonkamanda, 2022-04-09:
I added the usual "LD_LIBRARY_PATH=$HOME/.mujoco/mujoco200/bin" 

I didn't do things in a very scientific way so I might have changed other things that caused this but I am confident enough that it might be the one who did the fix, I first tried pip install mujoco,then pip install free-mujoco-py which both gave me the error and when I just used the open AI way with the path it worked.

saran-t, 2022-04-09:
Does `$HOME/.mujoco/mujoco200/bin` actually contain MuJoCo version 2.0.0, or one of our newer versions?

When you `pip install mujoco`, the package actually comes with its own MuJoCo library and it won't even try to look outside of its own package directory for any alternative copy of MuJoCo.

I would like to get to the bottom of this issue but I'll need your help since I cannot reproduce this on my side. Would you be able put your setup back into the configuration that caused the error and set a breakpoint in `dm_control/mujoco/engine.py` on the line where `wrapper.MjrContext` is being called, and print out the `render_context` object?

jianrenw, 2022-04-11:
I met the same problem, and it cannot be solved by the easy fix. Here is he print out of render_context: "<dm_control._render.glfw_renderer.GLFWContext object at 0x7f4ed3af43d0>"

saran-t, 2022-04-11:
In `_render/glfw_renderer.py` around Line 20 (right after the first two imports), could you please try inserting the following?

```
import ctypes
import ctypes.util
ctypes.CDLL(ctypes.utils.find_library('GL'), ctypes.RTLD_GLOBAL)
```

jianrenw, 2022-04-11:
Thank you for your reply. I think it works now. Just a minor typo: "ctypes.utils"->"ctypes.util"

saran-t, 2022-04-11:
Thank you, fixed the typo. I'll also try to "fix" MuJoCo itself for the next release.

Some more background on what's happening: our current GLAD setup assumes that users link their choice of GL dynamic library (`libOSMesa.so, `libEGL.so`, or `libGLX.so`) into their application at build time, and the GLAD load code assumes that one of `eglGetProcAddress`, `OSMesaGetProcAddress`, or `glXGetProcAddressARB` has already been loaded into the process' symbol table.

In the Python use case, this isn't necessarily true. One solution would be to modify the loading code to explicitly `dlopen` an OpenGL dynamic library if a `*GetProcAddress` function isn't already available.

In case anyone's interested, here's the relevant loading code that we're currently using on Linux.

```typedef void* (APIENTRYP PFNGLXGETPROCADDRESSPROC_PRIVATE)(const char*);
static PFNGLXGETPROCADDRESSPROC_PRIVATE gladGetProcAddressPtr;

static
int open_gl(void) {
  // We try to load a GetProcAddress symbol that's already been loaded
  // into the process. Since MuJoCo relies on user code to set up a working
  // OpenGL environment, one of these symbols should always be present.
  void* handle = dlopen(NULL, RTLD_NOW | RTLD_GLOBAL);
  if (handle) {
    // Try EGL.
    gladGetProcAddressPtr = (PFNGLXGETPROCADDRESSPROC_PRIVATE)dlsym(handle,
        "eglGetProcAddress");
    if (gladGetProcAddressPtr) { return 1; }

    // Try OSMesa.
    gladGetProcAddressPtr = (PFNGLXGETPROCADDRESSPROC_PRIVATE)dlsym(handle,
        "OSMesaGetProcAddress");
    if (gladGetProcAddressPtr) { return 1; }

    // Try GLX.
    gladGetProcAddressPtr = (PFNGLXGETPROCADDRESSPROC_PRIVATE)dlsym(handle,
        "glXGetProcAddressARB");
    if (gladGetProcAddressPtr) { return 1; }
  }

  return 0;```

saran-t, 2022-04-13:
[MuJoCo 2.1.5](https://github.com/deepmind/mujoco/releases/tag/2.1.5) fixes this issue.

closed 2022-04-13 by saran-t

AninditaDeb, 2022-12-08:
Hi,

 I have tried to install mujoco using the mujoco bindings rather than using mujoco-py exactly according to the instructions mentioned in this path https://github.com/deepmind/mujoco/blob/main/python/README.md, I have tried on windows, windows subsystem for linux and google colab as well , but every time I am coming up with a new error, this time after executing "bash make_sdist.sh" , it says - MUJOCO_PLUGIN_PATH environment variable is not set, can the experts from this community suggest how to resolve this?

[image]

saran-t, 2022-12-08:
@AninditaDeb Sorry about this, we need to update our documentation there.

When you unpack the prebuilt MuJoCo tarball, you should find a directory in there called `bin/mujoco_plugin` `MUJOCO_PLUGIN_PATH` to the path to that directory.

AninditaDeb, 2022-12-08:
@saran-t 
Thank you for your advise, I have tried to search this directory bin/mujoco_plugin, exactly where it is supposed to be present?

[image]

saran-t, 2022-12-08:
@AninditaDeb You said that you were following [these instructions](https://github.com/deepmind/mujoco/blob/main/python/README.md). Step 2 says that you should download [the binary release](https://github.com/deepmind/mujoco/releases/tag/2.3.1), which you should then unpack. You'll find `bin/mujoco_plugins` in there.

Having said all that, reading your comments again I'm under the impression that you just want to install `mujoco`. The [same instructions](https://github.com/deepmind/mujoco/blob/main/python/README.md) say that the recommended method is `pip install mujoco`. Does this not work for you?

AninditaDeb, 2022-12-09:
@saran-t 
I have successfully executed the below steps
1)Download the [latest binary release](https://github.com/deepmind/mujoco/releases) from GitHub
I have downloaded [**mujoco-2.3.1-linux-x86_64.tar.gz**] from and (https://github.com/deepmind/mujoco/releases) and kept this tar file inside a new directory mujoco_new on Ubuntu 20.4 LTS - the exact location is -/home/annie07/mujoco_new
1)git clone https://github.com/deepmind/mujoco.git
2)cd mujoco/python 
3)python3 -m venv /tmp/mujoco
4)source /tmp/mujoco/bin/activate
5)cd python
6)bash make_sdist.sh
On executing the above command i get this tar file mujoco-2.3.1.post1.tar.gz inside this directory - /home/annie07/mujoco/python/dist
However I fail to execute the below step

7)cd dist
   [image]

I am not sure exactly what I am missing

AninditaDeb, 2022-12-09:
I forgot to  mention when I execute MUJOCO_PATH=/home/annie07/mujoco_new pip install mujoco-2.3.1.post1.tar.gz, inside this directory /mujoco/python/dist , I get the above error

saran-t, 2022-12-09:
Have you tried actually just `pip install mujoco` ?

AninditaDeb, 2022-12-09:
Yes i did it as the first step which was basically the first step but I forgot to mention earlier.

saran-t, 2022-12-09:
What happened when you `pip install mujoco`? That's should be all that you need to do.

AninditaDeb, 2022-12-09:
@saran-t
Hi,
I believe I don't have to do those extra steps as you said just pip install mujoco worked 
But I was just trying Open AI GYM, I am not able to render , initially I didn't provide any value to in env = gym.make('Humanoid-v4')in that case no window pops up and then I set the  render_mode  as human but it still doesn't work and it fails with the below error
[image]
I believe its has something to do with  the glfw files
I know this issue might not be explicilty related to mujoco but can you please advise me exactly where to look at?

saran-t, 2022-12-09:
Looks like you're using this over SSH. GLFW doesn't play well over SSH since it relies on X11 (specifically glx) to render.

If the machine has an Nvidia GPU, your best bet is to set the environment variable `MUJOCO_GL=egl`. Otherwise, you can use OSMesa for software rendering (`apt install libosmesa6-dev` then set `MUJOCO_GL=osmesa`) but this will be very slow.

AninditaDeb, 2022-12-09:
@saran-t 
Hi,

I have few doubts regarding the above issue -

I am working on Ubuntu20.4 LTS  on windows 10 machine so as you mentioned above I executed this command (apt install libosmesa6-dev) but it seems its already installed -ss
[image]

Can you please tell me exactly where do I set this variable MUJOCO_GL=osmesa?

Also I was of the assumption that in Google Colab these installations will run smoothly, but after executing pip install mujoco when I run the below code
 import gym
env = gym.make('Humanoid-v4')
for i_episode in range(200):
  observation = env.reset()
  for t in range(100):
    env.render()
    print(observation)
    action = env.action_space.sample()
    observation, reward, done, info,  _ = env.step(action)
    if done:
      print("Episode finished after {} timesteps".format(t+1))
      break
env.close()
It gives me the error - You appear to be missing MuJoCo.  We expected to find the file here: /root/.mujoco/mujoco210 which I  believe this is not required as I have the most updated one 2.3.1.post1,
[image]
[image]

Is it possible to completely avoid the rendering issues, bypass them?

saran-t, 2022-12-09:
@AninditaDeb your question really has nothing to do with this issue's topic. It also has nothing to do with `dm_control`. You should've opened a new issue on https://github.com/deepmind/mujoco.

> I am working on Ubuntu20.4 LTS on windows 10 machine

We've never tested this setup, and I also don't have any expertise on how (or whether) WSL emulates X11. If you need rendering to work on Windows I reckon you'll have much better luck by just installing the Windows Python package rather than going through WSL.

> Can you please tell me exactly where do I set this variable MUJOCO_GL=osmesa?

`export MUJOCO_GL=osmesa`

> after executing pip install mujoco when I run the below code `import gym env = gym.make('Humanoid-v4')` ...  It gives me the error - You appear to be missing MuJoCo. We expected to find the file here: /root/.mujoco/mujoco210`

This is an issue with Gym, you should ask on the authors of that package.

> Is it possible to completely avoid the rendering issues, bypass them?

Again, this is a question for Gym authors. The MuJoCo bindings doesn't try to initialise the rendering stack unless someone asks it to (in this case, `gym`).

AninditaDeb, 2022-12-09:
@saran-t 
Hi ,
Thank you very much for your suggestions, I suppose I should raise this issue with the Gym authors.

sanky29, 2023-06-14:
hi,
i am still facing issue
`mujoco.FatalError: gladLoadGL error`

I have installed mujoco by
`pip install mujoco`
I have `mujoco-2.3.5` installed on linux with `python3.8.8`

sanky29, 2023-06-14:
Anyway, I resolved the error. So for those who are working on remote servers with the headless environment but have NVIDIA GPUs the following post may help:
[https://pytorch.org/rl/reference/generated/knowledge_base/MUJOCO_INSTALLATION.html](https://pytorch.org/rl/reference/generated/knowledge_base/MUJOCO_INSTALLATION.html)

xiaofeifan0727, 2023-10-31:
> Anyway, I resolved the error. So for those who are working on remote servers with the headless environment but have NVIDIA GPUs the following post may help: https://pytorch.org/rl/reference/generated/knowledge_base/MUJOCO_INSTALLATION.html

hi,i have the same problem, and your suggestion helps me solve it, Thanks.

pratik-ingle, 2024-05-14:
I am facing the same error with mujoco 3.1.1, while using it in singularity container on a HPC cluster. 

Traceback (most recent call last):
  File "/home/prin/sbatch_tests/train_clust.py", line 45, in <module>
    env = gym.make("soft_bot/SurfaceBot-v0", model_path=xml_path, framerate=60, vis=True)
  File "/usr/local/lib/python3.10/dist-packages/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
  File "/home/soft-bot/soft_bot/envs/manipulator.py", line 54, in __init__
    self.renderer = mujoco.Renderer(self.model, resolution[0], resolution[1])
  File "/usr/local/lib/python3.10/dist-packages/mujoco/renderer.py", line 83, in __init__
    self._mjr_context = _render.MjrContext(
mujoco.FatalError: gladLoadGL error

YechenZhou, 2024-07-20:
> The link doesn't work. I am facing the same error with mujoco 3.1.1, while using it in singularity container on a HPC cluster.
> 
> Traceback (most recent call last): File "/home/prin/sbatch_tests/train_clust.py", line 45, in env = gym.make("soft_bot/SurfaceBot-v0", model_path=xml_path, framerate=60, vis=True) File "/usr/local/lib/python3.10/dist-packages/gymnasium/envs/registration.py", line 802, in make env = env_creator(**env_spec_kwargs) File "/home/soft-bot/soft_bot/envs/manipulator.py", line 54, in **init** self.renderer = mujoco.Renderer(self.model, resolution[0], resolution[1]) File "/usr/local/lib/python3.10/dist-packages/mujoco/renderer.py", line 83, in **init** self._mjr_context = _render.MjrContext( mujoco.FatalError: gladLoadGL error

Have you solved it?

sanky29, 2024-08-05:
try this new link:
https://pytorch.org/rl/main/reference/generated/knowledge_base/MUJOCO_INSTALLATION.html

DengyingFu, 2025-05-29:
I set `export MUJOCO_GL=egl` then it works

yuyu5333, 2025-08-22:
On the GPU cloud server, when starting the service using a Docker container, this error still occurs. I have tried many measures for MUJOCO_GL, but none of them have been resolved.