google-deepmind/dm_control #284: How to define new task ?

jrabary, 2022-04-08:
We would like to simulate a custom robotic manipulation task environment and then use it in a reinforcement learning framework.
It is possible to use dm_control components to do that and is there any guidelines on how to proceed ?

kevinzakka, 2022-04-08:
I used the tasks in `manipulation`, `locomotion` and `suite` to learn the API. They're extremely well-written and should cover a lot of uses cases.

jrabary, 2022-04-09:
Thanks @kevinzakka! 
I find there is another way to create custom task in dm_robotics https://github.com/deepmind/dm_robotics/blob/main/py/moma/README.md
It uses dm_control as well. I’m wondering which is the best way to go ?

kevinzakka, 2022-04-09:
@jrabary The dm_robotics repo builds on top of dm_control and uses library-specific things like agentflow, base/subtask abstractions, etc. It has more features but also probably a bigger learning curve / hassle. I'd recommend you start with dm_control as it should cover more RL use cases.