google-deepmind/dm_control #286: Updating positions of sites during environment reset

mohakbhardwaj, 2022-04-10:
Hi, 
I have an environment where the robot needs to reach different goal positions that are randomized at every timestep. I am representing the goal as a site on the worldbody in the xml file for visualization purposes. However, while I update the `site_xpos` using  ` with physics.reset_context():
physics.named.data.site_xpos["goal"] = np.random.uniform(low,high)`
the location of the site does not get updated in the dm_viewer. Is there a way to update site position in the viewer or am I using the functionality incorrectly? Thanks!

yuvaltassa, 2022-04-10:
You want to change model.site_pos rather than data.site_xpos. xpos quantities get overwritten by kinematics.

Makes sense?

kevinzakka, 2022-04-10:
Adding to @yuvaltassa, if you have the site as an `mjcf.Element`, you can bind the `mjcf.Physics` object to it and query/change physical quantities much more easily. You can read about it [here](https://github.com/deepmind/dm_control/blob/master/dm_control/mjcf/physics.py#L549).

mohakbhardwaj, 2022-04-10:
> As an aside, if you have the site as an `mjcf.Element`, you can bind the `mjcf.Physics` object to it and query/change physical quantities much more easily. You can read about it [here](https://github.com/deepmind/dm_control/blob/master/dm_control/mjcf/physics.py#L549).

Yes, this worked for me. Thanks!