google-deepmind/dm_control #290: Create a manipulable entity based on STL mesh

jrabary, 2022-04-26:
I'm trying to build a custom manipulation task. I closely follow the construction of the Lift task in dm_control.

The model is defined as follow

```
<mujoco model="left_gear">
  <asset>
    <mesh file="meshes/Left_gear.stl" name="left_gear_mesh" scale="0.001 0.001 0.001"/>
    <material name="gear" reflectance="0.5"/>
  </asset>
  <worldbody>
    <body>
      <body name="object">
        <geom pos="0 0 0" name="left_gear" mesh="left_gear_mesh" type="mesh" solimp="0.998 0.998 0.001" solref="0.001 1" density="50" friction="0.95 0.3 0.1" material="gear" contype="0" conaffinity="0" group="1"/>
      </body>
    </body>
  </worldbody>
</mujoco>
```

The corresponding entity

```
class LeftGear(composer.Entity):
    def _build(self):
        self._mjcf_root = mjcf.from_path(_LEFT_GEAR_XML_PATH)

    @property
    def mjcf_model(self):
        return self._mjcf_root

    @property
    def geom(self):
        return self._mjcf_root.find("geom", "left_gear")
```

The task definition

```
class HelicalGear(composer.Task):
    
    def __init__(self, workspace: Workspace):
        obs_settings = observations.VISION
        self._arena = arenas.Standard()
        self._arm = robots.make_arm(obs_settings=obs_settings)
        self._hand = robots.make_hand(obs_settings=obs_settings)
        self._arm.attach(self._hand)
        self._arena.attach_offset(self._arm, [0, 0.4, 0])
        
        self._task_observables = cameras.add_camera_observables(self._arena, obs_settings, cameras.FRONT_CLOSE)
        
        self._tcp_initializer = initializers.ToolCenterPointInitializer(
            self._hand, 
            self._arm,
            position=distributions.Uniform(low=workspace.tcp_bbox.lower, high=workspace.tcp_bbox.upper),
            quaternion=workspaces.DOWN_QUATERNION
        )
        
        self._left_gear = left_gear.LeftGear()
        self._left_gear.geom.mass = left_gear.MASS
        self._arena.add_free_entity(self._left_gear)

        self._prop_placer = initializers.PropPlacer(
            props=[self._left_gear],
            position=distributions.Uniform(low=workspace.prop_bbox.lower, high=workspace.prop_bbox.upper),
            quaternion=workspaces.uniform_z_rotation,
            ignore_collisions=True,
            settle_physics=True
        )
        
    @property
    def root_entity(self):
        return self._arena
    
    @property
    def task_observables(self):
        return self._task_observables
    
    def get_reward(self, physics):
        return 0

    def initialize_episode(self, physics, random_state):
        self._hand.set_grasp(physics, close_factors=random_state.uniform())
        self._prop_placer(physics, random_state)
        self._tcp_initializer(physics, random_state)
        
        # # Compute the target height based on the initial height of the prop's
        # # center of mass after settling.
        # initial_prop_height = self._get_height_of_lowest_vertex(physics)
        # self._target_height = _DISTANCE_TO_LIFT + initial_prop_height
        # physics.bind(self._target_height_site).pos[2] = self.
```

At this point I'm able to view the scene inside the viewer but when I run the animation with a random policy the gear disappear from the scene. Also when I use the prop_placer the gear is not visible on the scene.

I certainly miss something but can't find out.

Any ideas ?

jrabary, 2022-04-26:
I get this warning

```
WARNING:absl:Failed to settle physics after 1 attempts of 2.0 seconds.
Last residual velocity=19.61999999999982 and acceleration=9.81.
This suggests your dynamics are unstable. Consider:
        1. Increasing `max_settle_physics_attempts`
        2. Increasing `max_settle_physics_time`
        3. Tuning your contact parameters or initial pose distributions.
```

I guess it is something to do with the contact parameters ?

kevinzakka, 2022-04-26:
@jrabary Can you attach a zip file with your code + assets and I can try taking a look?

jrabary, 2022-04-27:
Here is the code [control_ext.zip](https://github.com/deepmind/dm_control/files/8569777/control_ext.zip)

kevinzakka, 2022-04-27:
Hi @jrabary , your issue is that your mesh has `contype="0" conaffinity="0"`. Those disable collision checking. Just remove them and it should work.

yuvaltassa, 2022-04-27:
Ha! 
Great catch @kevinzakka. 
I did browse through the OP's code above, but forgot to actually drag that horizontal scroll bar aaall the way to the right :stuck_out_tongue:

closed 2022-04-27 by yuvaltassa