google-deepmind/dm_control #293: how to load customized model from xml?

jinfagang, 2022-04-28:
Hi, how to load customized model from xml?

kevinzakka, 2022-04-28:
Can you be more specific? Do you mean you have an XML file path and you want to load the model?

jinfagang, 2022-04-28:
@kevinzakka Hi, I just digged into issues and found a solution. But I found an issue about dm_control or maybe mujoco, Why does mujoco read xml file always need absolute path? that was weired.

it's maybe not mujoco, it's dm ctonrl issue:

```
from dm_control.suite import common
common.read_model(MODEL_XML_PATH), common.ASSETS

```

it will read model in asolute path, and all path inside xml, should be absolute, otherwise cause not found error. Why??

kevinzakka, 2022-04-28:
You're using `suite` specific methods here. In general, you can just read an XML file using `mujoco` itself:

```python
import mujoco
mujoco.MjModel.from_xml_path(MODEL_XML_PATH)
```

and it doesn't require absolute paths for the meshes or textures.

jinfagang, 2022-04-29:
thank u, I just fixed it.

closed 2022-04-29 by jinfagang