google-deepmind/dm_control #294: how to read model with assets in dm_control?

jinfagang, 2022-04-28:
I try to load a custom model:

```
THIS_DIR = os.path.dirname(__file__)
# MODEL_XML_PATH = '/home//mujoco/model/humanoid/humanoid.xml'
# MODEL_XML_PATH = os.path.join(THIS_DIR, './models/humanoid_h36m/humanoid_h36m_v5.xml') 
MODEL_XML_PATH = 'models/mug/mug.xml'

def get_model_and_assets():
    """Returns a tuple containing the model XML string and a dict of assets."""
    # return common.read_model(MODEL_XML_PATH), common.ASSETS
    # return resources.GetResource(MODEL_XML_PATH), common.ASSETS
    return resources.GetResource(MODEL_XML_PATH), {}

```

It always gives me this error:

```
python3.9/site-packages/dm_control/mujoco/wrapper/core.py", line 187, in _get_model_ptr_from_xml
    ptr = mujoco.MjModel.from_xml_string(xml_string, assets or {})
ValueError: Error: could not parse OBJ file 'mug.obj': 
Cannot open file [mug.obj]

Object name = mug, id = 0, line = 10, column = -1

```

I know it might because I need also assets to specific, but I don't know how to specific, is that assets is a dir which contains all my assets? (like stl, png, mesh etc.)

Any detailed examples? thanks!

kevinzakka, 2022-04-29:
Hi @jinfagang, here's a quick example for how to load the mug and its assets.

```python
from dm_control import mujoco
from PIL import Image

MODEL_DETAILS = {
    "xml": '/Users/kevin/dev/repos/mujoco/model/mug/mug.xml',
    "assets": {
        "mug.obj": '/Users/kevin/dev/repos/mujoco/model/mug/mug.obj',
        "mug.png": '/Users/kevin/dev/repos/mujoco/model/mug/mug.png'
    }
}

def get_model_and_assets():
    with open(MODEL_DETAILS["xml"], 'rb') as f:
        xml = f.read()
    assets = {}
    for asset_name, asset_path in MODEL_DETAILS["assets"].items():
        with open(asset_path, 'rb') as f:
            assets[asset_name] = f.read()
    return xml, assets

physics = mujoco.Physics.from_xml_string(*get_model_and_assets())
Image.fromarray(physics.render(480, 640)).show()
```

kevinzakka, 2022-04-29:
@jinfagang You shouldn't really be reading from an XML string if you have the path to the model. You can read in the entire model in one line with:

```python
from dm_control import mujoco
from PIL import Image

physics = mujoco.Physics.from_xml_path("/Users/kevin/dev/repos/mujoco/model/mug/mug.xml")
Image.fromarray(physics.render()).show()
```
and now you don't have to read the assets separately.

closed 2022-05-07 by jinfagang