google-deepmind/dm_control #298: CMUwalker can not be run

jinfagang, 2022-05-07:
Hi, the cmuwalker example from ipythonnotebook I got error on my side:

```
dm_control viewer intercepted an environment error.
Original message: operands could not be broadcast together with shapes (3,) (2,)
```

Here is the main func:

```
def cmu_walker():
    walker = cmu_humanoid.CMUHumanoidPositionControlledV2020(
        observable_options={"egocentric_camera": dict(enabled=True)}
    )
    arena = corridor_arenas.WallsCorridor(
        wall_gap=3,
        wall_width=distributions.Uniform(2, 3),
        wall_height=distributions.Uniform(0.5, 1.5),
        corridor_width=4,
        corridor_length=30,
    )
    task = corridor_tasks.RunThroughCorridor(
        walker=walker,
        arena=arena,
        walker_spawn_position=(0.5, 0),
        target_velocity=3.0,
        physics_timestep=0.005,
        control_timestep=0.03,
    )
    env = composer.Environment(
        task=task,
        time_limit=10,
        random_state=np.random.RandomState(42),
        strip_singleton_obs_buffer_dim=True,
    )

    action_spec = env.action_spec()
    print(action_spec.shape)

    def random_policy(time_step):
        del time_step
        rp = np.random.uniform(
            low=action_spec.minimum, high=action_spec.maximum, size=action_spec.shape
        )
        print(rp.shape)
        return rp

    viewer.launch(env, policy=random_policy)
```

I don't know why the (3,) (2,) come frome, the control policy should be just (56)

kevinzakka, 2022-05-07:
`shift_pose` expects a position of length 3 (x, y and z), but your `walker_spawn_position=(0.5, 0)` only specifies x, y. Add a z-dimension and it should work.

closed 2022-05-07 by jinfagang

jinfagang, 2022-05-07:
@kevinzakka thanks! you saved my day!

may I ask another question? am pretty new to mujoco, I wanna ask, for my customized humanoid model, it's action space is 52, what's the every value means inside these 52 values？
I want using human36m GT action data to driven the model, how should I able to control the model do the exact same action like human36m? thanks aton.