google-deepmind/dm_control #299: How to change mujoco model parameters to pytorch tensors?

ZJP-c4cld, 2022-05-10:
I want to construct a neural network in pytorch which outputs appropriate values of mujoco model parameters. But this requires that mujoco model parameters are converted into pytorch tensors and mujoco environments run well with these converted model parameters. I don't know how to achieve this goal. Anyone know how to do that?

vmoens, 2022-06-16:
Hi, @ZJP-c4cld, that seems like an interesting project.
Can you detail a bit more what you'd like to do?
What you should do is overwrite the `data` associated with your model (e.g. [here](https://github.com/deepmind/dm_control/blob/41d0c7383153f9ca6c12f8e865ef5e73a98759bd/dm_control/mujoco/engine.py#L160-L168))
but the gradients won't pass through the simulator (for that you could consider Brax)

ZJP-c4cld, 2022-06-19:
Hello,@vmoens. Thanks for your attention to this topic. My detailed explaination is as follows:

My research interest is robustness of RL algorithms to environment parameters. I want to modify currents RL algorithms to make them achieve good performance when they are tested in environments with unfamilar parameters. (For example, an agent is trained in Cartpole environment with 1m pole. I want it achieve good performance in Cartpole environment with 3m pole.)

To achieve this goal, I want to get the relationship between model parameter values and RL algorithm's performance (reward). As a result, I want to get the gradient of reward with respect to the model parameters.

Mujoco simulator has applied in my experiments. But Mujoco simulator is not implemented by pure python. So I cannot get the the gradient of reward with respect to the model parameters. So I want to try convert mujoco model parameters into pytorch tensors and make mujoco environments run well with these converted model parameters.

ZJP-c4cld, 2022-06-22:
@vmoens Thank you very much ^_^

yuvaltassa, 2022-06-23:
@vmoens, not entirely what the OP was asking, but back-propagating through the simulation w.r.t state and control is now possible since [228264c](https://github.com/deepmind/mujoco/commit/228264c92b17975e9e05874a5e273f2fbaa62163).

[This function](https://github.com/deepmind/mujoco/blob/main/include/mujoco/mujoco.h#L1086) provides access to an efficient implementation of finite differenced derivatives ([implementation](https://github.com/deepmind/mujoco/blob/main/src/engine/engine_derivative.c#L1031), [basic test](https://github.com/deepmind/mujoco/blob/main/test/engine/engine_derivative_test.cc#L320)). Note that MuJoCo's contact model is smooth so the dynamics can be made perfectly differentiable.

At some point we intend to replace this with fully analytic derivatives, but that will take time. The finite-differenced derivatives are reasonably fast and can be used for trajectory optimisation, including in real-time (e.g. [paper](https://homes.cs.washington.edu/~todorov/papers/TassaICRA14.pdf), [video](https://www.youtube.com/watch?v=8yFBe0VeBoA)).

Regardless of the analytic/FD implementation, we will soon release more API tools and demos using Jacobians of various top level functions.

Regarding @ZJP-c4cld's OP, I doubt that any existing simulator will give you analytical derivatives w.r.t model parameters, though both Brax and DiffTaiChi are worth a look in that regard. However as all of RL is basically a collection of algorithms estimating the value gradient w.r.t _something_ , these same algorithms can be directly applied to to estimate gradients w.r.t model parameters, in MuJoCo and elsewhere, no _actual_ gradients required 🙂

closed 2022-06-23 by yuvaltassa

vmoens, 2022-06-23:
Wow, This is an amazing feature!
Sorry I missed that!

vmoens, 2022-06-23:
@yuvaltassa any quick recipe to use that feature with dm_control physic engine?

yuvaltassa, 2022-06-24:
Not yet.

To be clear, because `dm_control` is pinned to _released_ MuJoCo versions, it is not even possible to do so in principle ATM.

Soon! 🙂
(Like, a couple of weeks)

ZJP-c4cld, 2022-06-24:
@yuvaltassa Thank you for your selfless help