google-deepmind/dm_control #302: python: /builds/florianrhiem/pyGLFW/glfw-3.3.7/src/posix_thread.c:64: _glfwPlatformGetTls: Assertion `tls->posix.allocated == 1' failed.

DexiongYung, 2022-05-14:
Getting this error 

python: /builds/florianrhiem/pyGLFW/glfw-3.3.7/src/posix_thread.c:64: _glfwPlatformGetTls: Assertion `tls->posix.allocated == 1' failed.

when it gets to the `env.render()` function. Not sure what to do.

```
import dmc2gym

env = dmc2gym.make(domain_name='point_mass', task_name='easy', seed=1)

done = False
obs = env.reset()
while not done:
  action = env.action_space.sample()
  obs, reward, done, info = env.step(action)
  rgb_obs = env.render(mode='rgb_array')
```

kevinzakka, 2022-05-14:
Is your `dm_control` version compatible with your MuJoCo version? Looks like you are using 200 but the newest `dm_control` is pegged to `2.1.5`

DexiongYung, 2022-05-14:
solution I found online was typing in these commands

```
export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so 
xvfb-run -a -s "-screen 0 1400x900x24" bash
```

closed 2022-05-14 by DexiongYung