google-deepmind/dm_control #303: Path issues with Mujoco

oofmeister27, 2022-05-17:
I installed dm_control using pip and I was attempting to try it out with [this repository](https://github.com/danijar/dreamerv2). When I ran this command: `python3 dreamerv2/train.py --logdir ~/logdir/dmc_walker_walk/dreamerv2/1 \
  --configs dmc_vision --task dmc_walker_walk`

I get this error: `OSError: Cannot find MuJoCo library at /home/*home folder*/.mujoco/mujoco200/bin/libmujoco200.dylib.`

I found this [issue](https://github.com/deepmind/dm_control/issues/122)  and tried following the steps outlined in [this](https://github.com/deepmind/dm_control/issues/122#issuecomment-568791401) post, but for some reason I still get that same error. 

Would appreciate if anyone could take a look at let me know what the issue is.

kevinzakka, 2022-05-17:
You probably installed a cached version. Try:

```bash
pip install --upgrade dm_control
```

and double check it installed version [1.0.2](https://github.com/deepmind/dm_control/blob/main/setup.py#L176).

oofmeister27, 2022-05-18:
Hm, I ran that command and it just gave me multiple lines of "Requirement
already satisfied". I did try to do ```pip install --upgrade
dm_control=1.0.2``` and for some reason, it gave me this error: ```ERROR:
Could not find a version that satisfies the requirement dm_control==1.0.2
(from versions: 0.0.286587932, 0.0.286955599, 0.0.288398964, 0.0.288483845,
0.0.295778102, 0.0.300771433, 0.0.312466143, 0.0.318037100, 0.0.318066097,
0.0.319497192)
ERROR: No matching distribution found for dm_control==1.0.2```

I believe I may have an older dm_control installed, but not sure why its
not able to install 1.0.2

kevinzakka, 2022-05-18:
@oofmeister27 what OS are you using?

oofmeister27, 2022-05-18:
@kevinzakka I am using Ubuntu 22.04

saran-t, 2022-05-18:
You may be able to get more useful debugging info by running `pip install -vv --no-deps dm_control==1.0.2`

saran-t, 2022-05-18:
Wait, are you using Python 2.7? Looks like the last version that `pip` reports (0.0.319497192) corresponds to the last version on PyPI with Python 2 support.

We no longer support Python 2.

oofmeister27, 2022-05-18:
@saran-t I am using python 3.10.4. 

@saran-t @kevinzakka I ran this command ```pip install -vv --no-deps dm_control==1.0.2``` as you suggested and here's the output (its pretty long): https://pastebin.com/aumJaAwh

From what I see, I think it is something to do with the python version since its not able to find dm_control 1.0.2. However, if I run python3 --version it does return 3.10.4 so I'm not sure what's wrong.

saran-t, 2022-05-18:
Could you please try `pip install --upgrade --ignore-requires-python dm_control==1.0.2` ?

oofmeister27, 2022-05-19:
@saran-t Hm, I ran that command and got this error: https://pastebin.com/ZX4CvUkp

saran-t, 2022-05-19:
`labmaze` (which is one of `dm_control`'s dependencies) currently [does not ship a prebuilt wheel for 3.10](https://pypi.org/project/labmaze/1.0.5/#files).

If you [install Bazel](https://bazel.build/install) your install should succeed. Alternatively, I will look into pushing out `labmaze` wheels for 3.10, but I can't promise any specific timeframe at the moment, unfortunately.

closed 2022-05-19 by saran-t

oofmeister27, 2022-05-21:
Thanks, I got it to work now. @saran-t @kevinzakka

commit 3e1736d7 (2022-05-19): Remove maximum Python version requirement for dm_control.

Fixes #303.

PiperOrigin-RevId: 449711039
Change-Id: Ibc1f3173029d248374a58222ec2d1c0e5212eeee