google-deepmind/dm_control #304: CentOS7 Cannot initialize a headless EGL display.

945716994, 2022-05-20:
File "src/train.py", line 14, in <module>
    from env import make_env
  File "/home/wjg/code/src/env.py", line 5, in <module>
    from dm_control import suite
  File "/home/wjg/anaconda3/envs/tdmpc/lib/python3.8/site-packages/dm_control/suite/__init__.py", line 24, in <module>
    from dm_control.suite import acrobot
  File "/home/wjg/anaconda3/envs/tdmpc/lib/python3.8/site-packages/dm_control/suite/acrobot.py", line 20, in <module>
    from dm_control import mujoco
  File "/home/wjg/anaconda3/envs/tdmpc/lib/python3.8/site-packages/dm_control/mujoco/__init__.py", line 18, in <module>
    from dm_control.mujoco.engine import action_spec
  File "/home/wjg/anaconda3/envs/tdmpc/lib/python3.8/site-packages/dm_control/mujoco/engine.py", line 41, in <module>
    from dm_control import _render
  File "/home/wjg/anaconda3/envs/tdmpc/lib/python3.8/site-packages/dm_control/_render/__init__.py", line 67, in <module>
    Renderer = import_func()  # pylint: disable=invalid-name
  File "/home/wjg/anaconda3/envs/tdmpc/lib/python3.8/site-packages/dm_control/_render/__init__.py", line 36, in _import_egl
    from dm_control._render.pyopengl.egl_renderer import EGLContext
  File "/home/wjg/anaconda3/envs/tdmpc/lib/python3.8/site-packages/dm_control/_render/pyopengl/egl_renderer.py", line 75, in <module>
    raise ImportError('Cannot initialize a headless EGL display.')
ImportError: Cannot initialize a headless EGL display.

**Is this due to the system?**

945716994, 2022-05-20:
I got it!

945716994, 2022-05-20:
I can set the os.environ['MUJOCO_GL'] = 'glfw' ,but it's so slowly.
Did there have other method to acclerate ?

saran-t, 2022-05-20:
Can you tell us what GPU is present on your system?

claudiofantacci, 2022-05-20:
Hi @945716994, thanks for reaching out!

> NVIDIA A100

Is it a server or workstation grade A100?
Also, is it in [Multi-Instance GPU](https://docs.nvidia.com/datacenter/tesla/mig-user-guide/) (MIG) mode?

945716994, 2022-05-21:
> Hi @945716994, thanks for reaching out!
> 
> > NVIDIA A100
> 
> Is it a server or workstation grade A100? Also, is it in [Multi-Instance GPU](https://docs.nvidia.com/datacenter/tesla/mig-user-guide/) (MIG) mode?

Hi @claudiofantacci ,is a server and not in MIG,we have four A100 on server.

Kallin199, 2022-07-20:
Hi, @945716994 I also encountered this problem in ubuntu, how did you solve it later, can you tell me, thank you for your early reply.

baixianger, 2023-05-28:
I have the same problem, I was using my school's service featured with A100. And I found I cannot install the package `egl` and `osmesa` needed below.

```
sudo apt-get install libnvidia-gl-440-server
```
Because it's an LFS system, I cannot install any packages outside the module loading system.

Do you know how I can tackle this problem? I really appreciate any help you can provide.