google-deepmind/dm_control #307: Headless display?

slerman12, 2022-06-02:
I have 8 GPUs on a node and I guess only one of them has a head. So when I run a second job on them, I get a headless display error. The first job works fine.

Following that, I tried loading "glew" and now I'm getting this output:

```
[2022-06-01 23:18:45,712][HYDRA] MUJOCO_GL is not set, so an OpenGL backend will be chosen automatically.
[2022-06-01 23:18:45,786][HYDRA] Successfully imported OpenGL backend: glfw
[2022-06-01 23:18:46,023][HYDRA] MuJoCo library version is: 210
python3: /builds/florianrhiem/pyGLFW/glfw-3.3.4/src/posix_thread.c:64: _glfwPlatformGetTls: Assertion `tls->posix.allocated == 1' failed.
/var/spool/slurm/job11038422/slurm_script: line 25: 100365 Aborted 
```

vmoens, 2022-06-16:
Hi @slerman12 have you tried setting 

```
MUJOCO_GL=egl python my_script.py
```
?

slerman12, 2022-06-16:
Yes, I'm trying that. Together with MKL_SERVICE_FORCE_INTEL, not sure what that does.

slerman12, 2022-06-21:
Only on the GPU that has the head, not on the others. Is it supposed to be compatible with "headless displays"?

closed 2022-06-21 by slerman12