google-deepmind/dm_control #310: Noisy or unreadable images when rendering

vmoens, 2022-06-17:
Hi!

In some instances (embodied algos in my case) the new mujoco rendering gives unreadable images after a little while, e.g.:
[image]

This occurs after a little while, i.e. the first images rendered are perfectly fine. 
I tried to narrow it down to a minimal reproductible example but I can't find a way to do it (sorry about that!)
When using the old bindings (mujoco-py and such) thiis issue disappears. 

I'm using `MUJOCO_GL=egl` and have installed glew in my conda (working on a cluster where I have no sudo access).

I'm working with either G100 or A100 GPUs, and using them for training and rendering. 
Also to mention: I'm running a bunch of envs in parallel (not multithrerad but multiprocessing) for fast collection of data.

Here is my conda env

```
# packages in environment at /fsx/users/vmoens/conda/envs/rl4:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
_tflow_select             2.3.0                       mkl
absl-py                   1.0.0              pyhd8ed1ab_0    conda-forge
aiohttp                   3.8.1            py39hb9d737c_1    conda-forge
aiosignal                 1.2.0              pyhd8ed1ab_0    conda-forge
ale-py                    0.7.5                    pypi_0    pypi
alsa-lib                  1.2.6.1              h7f98852_0    conda-forge
anyio                     3.6.1                    pypi_0    pypi
aom                       3.3.0                h27087fc_1    conda-forge
argon2-cffi               21.3.0             pyhd8ed1ab_0    conda-forge
argon2-cffi-bindings      21.2.0           py39hb9d737c_2    conda-forge
astor                     0.8.1              pyh9f0ad1d_0    conda-forge
asttokens                 2.0.5              pyhd8ed1ab_0    conda-forge
astunparse                1.6.3              pyhd8ed1ab_0    conda-forge
async-timeout             4.0.2              pyhd8ed1ab_0    conda-forge
atari-py                  0.2.9                    pypi_0    pypi
attr                      2.5.1                h166bdaf_0    conda-forge
attrs                     21.4.0             pyhd8ed1ab_0    conda-forge
autorom                   0.4.2                    pypi_0    pypi
autorom-accept-rom-license 0.4.2                    pypi_0    pypi
babel                     2.10.1                   pypi_0    pypi
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
beautifulsoup4            4.11.1             pyha770c72_0    conda-forge
blas                      1.0                         mkl
bleach                    5.0.0              pyhd8ed1ab_0    conda-forge
blinker                   1.4                        py_1    conda-forge
brotlipy                  0.7.0           py39hb9d737c_1004    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2022.5.18.1          ha878542_0    conda-forge
cachetools                5.2.0                    pypi_0    pypi
certifi                   2022.5.18.1              pypi_0    pypi
cffi                      1.15.0           py39h4bc2ebd_0    conda-forge
charset-normalizer        2.0.12             pyhd8ed1ab_0    conda-forge
click                     8.1.3            py39hf3d152e_0    conda-forge
cloudpickle               1.2.2                    pypi_0    pypi
configargparse            1.5.3                    pypi_0    pypi
cryptography              37.0.1           py39h9ce1e76_0
cudatoolkit               11.3.1               h2bc3f7f_2
cycler                    0.11.0                   pypi_0    pypi
cython                    0.29.30                  pypi_0    pypi
dbus                      1.13.6               h5008d03_3    conda-forge
debugpy                   1.6.0            py39h5a03fae_0    conda-forge
decorator                 4.4.2                    pypi_0    pypi
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
dm-control                1.0.3.post1              pypi_0    pypi
dm-env                    1.5                      pypi_0    pypi
[... 281 lines cut ...]
xorg-glproto              1.4.17            h7f98852_1002    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libx11               1.7.2                h7f98852_0    conda-forge
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxcursor           1.2.0                h7f98852_0    conda-forge
xorg-libxdamage           1.1.5                h7f98852_1    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h7f98852_1    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxinerama          1.1.4             h9c3ff4c_1001    conda-forge
xorg-libxrandr            1.5.2                h7f98852_1    conda-forge
xorg-libxrender           0.9.10            h7f98852_1003    conda-forge
xorg-randrproto           1.5.0             h7f98852_1001    conda-forge
xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
xorg-util-macros          1.19.3               h7f98852_0    conda-forge
xorg-xextproto            7.3.0             h7f98852_1002    conda-forge
xorg-xf86vidmodeproto     2.3.1             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
yarl                      1.7.2            py39hb9d737c_2    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zipp                      3.8.0              pyhd8ed1ab_0    conda-forge
zlib                      1.2.12               h166bdaf_0    conda-forge
zstd                      1.5.2                h8a70e8d_1    conda-forge
```

saran-t, 2022-06-21:
Can you please try running with `DISABLE_RENDER_THREAD_OFFLOADING=1` (environment variable)?

ikostrikov, 2022-06-23:
@saran-t `DISABLE_RENDER_THREAD_OFFLOADING=1` doesn't resolve the problem for me either.

saran-t, 2022-06-23:
Can I please have a minimal repro code that I can run on my side?

saran-t, 2022-08-26:
@vmoens @ikostrikov  Gentle nudge on the request for minimal repro above. We'd like to try to get to the bottom of this.

vmoens, 2022-08-26:
Hi @saran-t 
I've been trying hard to reproduce this but it seems to only happen after the code reaches a certain level of complexity (e.g. gpus are used for training and rendering, etc.)
Would it be ok if I point you to a specific commit on [torchrl](https://github.com/facebookresearch/rl), give you the precise conda env setting, the machine config etc for you to reproduce?
It's going to be a bit messy but at least it's something!

saran-t, 2022-08-26:
If it's consistently reproducible, a messy repro case will be better than not having one at all, so please do give us that anyway.

Also, are you saying with like-for-like experiment complexity level, `mujoco-py` rendering does not break in the same way?

vmoens, 2022-08-26:
Here's one
[0e88eac27f1d01bfa1d260d52c051ab5fe514859](https://github.com/nicolas-dufour/torchrl/commit/0e88eac27f1d01bfa1d260d52c051ab5fe514859)

Here's the command line

```
conda create -n mbrl_dmcontrol3 python=3.10
conda activate mbrl_dmcontrol3
pip install dm_control
module load cuda/11.6 nccl/2.12.7-cuda.11.6 nccl_efa/1.15.1-nccl.2.12.7-cuda.11.6
pip3 install torch torchvision torchaudio --extra-index-url https://download.pytorch.org/whl/cu116
pip install functorch
pip install hydra-core
# from torchrl root:
python setup.py develop
cd examples/dreamer/
EGL_DEVICE_ID=2 MUJOCO_GL=egl CHECK_IMAGES=1 srun -p train --gpus-per-node 3 -c 32 python dreamer.py frame_skip=2 init_env_steps=10000 logger=csv
```

The [`CHECK_IMAGES=1`](https://github.com/nicolas-dufour/torchrl/blob/0e88eac27f1d01bfa1d260d52c051ab5fe514859/torchrl/data/tensor_specs.py#L237) will make sure an error is raise as soon as an image is more than half black or white (ie render has collapsed)

You should see an error like this during the first test rollout:
```
Traceback (most recent call last):
  File "/fsx/users/vmoens/work/rl_mb/examples/dreamer/dreamer.py", line 411, in main
    call_record(logger, record, collected_frames, sampled_tensordict_save, stats, model_based_env, actor_model, cfg)
  File "/fsx/users/vmoens/conda/envs/mbrl_dmcontrol3/lib/python3.10/site-packages/torch/autograd/grad_mode.py", line 27, in decorate_context
    return func(*args, **kwargs)
  File "/fsx/users/vmoens/work/rl_mb/examples/dreamer/dreamer.py", line 132, in call_record
    td_record = record(None)
  File "/fsx/users/vmoens/conda/envs/mbrl_dmcontrol3/lib/python3.10/site-packages/torch/autograd/grad_mode.py", line 27, in decorate_context
    return func(*args, **kwargs)
  File "/fsx/users/vmoens/work/rl_mb/torchrl/trainers/trainers.py", line 907, in __call__
    td_record = self.recorder.rollout(
  File "/fsx/users/vmoens/work/rl_mb/torchrl/envs/common.py", line 503, in rollout
    tensordict = self.reset()
  File "/fsx/users/vmoens/work/rl_mb/torchrl/envs/common.py", line 346, in reset
    tensordict_reset = self._reset(tensordict, **kwargs)
  File "/fsx/users/vmoens/work/rl_mb/torchrl/envs/transforms/transforms.py", line 403, in _reset
    out_tensordict = self.base_env.reset(execute_step=False, **kwargs)
  File "/fsx/users/vmoens/work/rl_mb/torchrl/envs/common.py", line 346, in reset
    tensordict_reset = self._reset(tensordict, **kwargs)
  File "/fsx/users/vmoens/work/rl_mb/torchrl/envs/gym_like.py", line 122, in _reset
    source=self._read_obs(obs),
  File "/fsx/users/vmoens/work/rl_mb/torchrl/envs/gym_like.py", line 136, in _read_obs
    observations = self.observation_spec.encode(observations)
  File "/fsx/users/vmoens/work/rl_mb/torchrl/data/tensor_specs.py", line 1107, in encode
    out[key] = self[key].encode(item)
  File "/fsx/users/vmoens/work/rl_mb/torchrl/data/tensor_specs.py", line 243, in encode
    assert v < 0.5, f"numpy: {val.shape}"
AssertionError: numpy: (240, 320, 3)
```

saran-t, 2022-08-26:
Please point me to where the rendering context is set up and where the multiprocessing occurs.

vmoens, 2022-08-27:
> If it's consistently reproducible, a messy repro case will be better than not having one at all, so please do give us that anyway.

got it!

> Also, are you saying with like-for-like experiment complexity level, `mujoco-py` rendering does not break in the same way?

let me rephrase: with one library where we used to rely on mujoco-py but switched to the new mujoco bindings, we have seen this issue appearing.
I ran the following experiment using an old version of dm_control with torchrl and the issue disappears.
Here's the setup

torchrl commit: [056699bd214937400c5cc7722669e7819a93bc1e](https://github.com/nicolas-dufour/torchrl/commit/056699bd214937400c5cc7722669e7819a93bc1e)

Setup:
```
conda create -n mbrl_olddmc python=3.9
conda activate mbrl_olddmc
pip install mujoco_py
pip install dm-control==0.0.403778684  # works with mujoco 210
pip3 install torch torchvision torchaudio --extra-index-url https://download.pytorch.org/whl/cu116
pip install functorch
pip install hydra-core
cd path/to/torchrl
python setup.py develop

conda env config vars set MJLIB_PATH=/data/home/vmoens/.mujoco/mujoco210/bin/libmujoco210.so LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/data/home/vmoens/.mujoco/mujoco210/bin MUJOCO_GL=egl PYOPENGL_PLATFORM=egl MUJOCO_PY_MUJOCO_PATH=/data/home/vmoens/.mujo
co/mujoco210
conda deactivate && conda activate mbrl_olddmc

```

Command:
```
EGL_DEVICE_ID=2 MUJOCO_GL=egl CHECK_IMAGES=1 srun -p train --gpus-per-node 3 -c 32 python dreamer.py frame_skip=2 init_env_steps=10000 logger=csv env_per_collector=1 num_workers=1
```

Importantly:
- this code immediately fails on the env where I have installed the new dm_control version (with new bindings)
- this setup will not use any parallelism for rendering (a single process is launched), see `env_per_collector=1 num_workers=1 async_collection=False` which tell our trainer to collect data on the same process where the training occurs.

For rendering, we use [dm_control pixels wrapper](https://github.com/nicolas-dufour/torchrl/blob/056699bd214937400c5cc7722669e7819a93bc1e/torchrl/envs/libs/dm_control.py#L155-L159).
When executing a [step](https://github.com/nicolas-dufour/torchrl/blob/056699bd214937400c5cc7722669e7819a93bc1e/torchrl/envs/gym_like.py#L99) we [create a torch. Tensor from the numpy array and send it on device if needed](https://github.com/nicolas-dufour/torchrl/blob/056699bd214937400c5cc7722669e7819a93bc1e/torchrl/data/tensor_specs.py#L259).

In the example script I gave here above, we first run a random rollout in the environment to get statistics about the observation. To do that, we have a [function](https://github.com/nicolas-dufour/torchrl/blob/056699bd214937400c5cc7722669e7819a93bc1e/torchrl/trainers/helpers/envs.py#L351) that creates an environment instance, runs the rollout and calculates the stats.
Then we run [another random rollout](https://github.com/nicolas-dufour/torchrl/blob/056699bd214937400c5cc7722669e7819a93bc1e/torchrl/trainers/helpers/models.py#L1164) to get data to pass to the model (to initialize it): we have lazy layers that take the right shape once they see real data. _In this example_, that's where the issue happens (not event during training).

saran-t, 2022-09-05:
I'm having trouble running `python dreamer.py frame_skip=2 init_env_steps=10000 logger=csv` on my machine.

Could you please make a repro script that just runs the dm_control environment without any agent in the loop, preferably without any dependency on Torch?

Note also that I don't have access to a SLURM cluster and I need to repro this on a local machine.

saran-t, 2022-09-05:
OK, I have this running. I have zero familiarity with this code, but it seems that Hydra is creating some sort of default `cfg` and is forcing `cfg.collector_devices` to be `['cuda:1', 'cuda:1']`. On my machine with only a single GPU, this causes an "invalid ordinal" CUDA error.

I had to go into `torchrl/trainers/helpers/envs.py` and manually override `device` to `'cuda:0'` which allows the script to run. However, now everything runs just fine and I cannot actually trigger the error.

vmoens, 2022-09-05:
Let me write a single-gpu example for you

saran-t, 2022-09-05:
I've managed to trigger the error. Still investigating, but it looks like something is copying the rendering context objects in Python, which isn't a supported operation.

saran-t, 2022-09-05:
@vmoens Could you please try https://github.com/saran-t/dm_control/pull/1 and see if it fixes your issue?

vmoens, 2022-09-05:
It is running in a much more stable way than it used to. No noisy pixel, and runs that used to collapse after a couple of iterations are now running smoothly. 
For me this can be considered as closed.
Thanks so much for your help @saran-t! This is amazing

closed 2022-09-05 by vmoens

saran-t, 2022-09-05:
I'll have this fixed in our 1.0.6 release later this week.

saran-t, 2022-09-07:
This should now be fixed in version 1.0.6.