google-deepmind/dm_control #317: Unexpected interaction between between dm_control and JAX

dyth, 2022-07-07:
The order of importing `jax` and `dm_control` has a large effect on FPS.
I'm using `dm-control==1.0.3`, `jax==0.3.1` and `jaxlib==0.3.0+cuda11.cudnn82`

The script below reproduces the issue, with code adapted from https://github.com/ikostrikov/jaxrl/tree/main/jaxrl/wrappers

What should be the correct order of importing the libraries?

```
fast = True
if fast:
    from dm_control import suite
    import jax
else:
    import jax
    from dm_control import suite

from dm_env import specs
import numpy as np
from typing import Dict, Optional, OrderedDict
import copy
import gym
from gym import core, spaces

def dmc_spec2gym_space(spec):
    if isinstance(spec, OrderedDict):
        spec = copy.copy(spec)
        for k, v in spec.items():
            spec[k] = dmc_spec2gym_space(v)
        return spaces.Dict(spec)
    elif isinstance(spec, specs.BoundedArray):
        return spaces.Box(low=spec.minimum,
                          high=spec.maximum,
                          shape=spec.shape,
                          dtype=spec.dtype)
    elif isinstance(spec, specs.Array):
        return spaces.Box(low=-float('inf'),
                          high=float('inf'),
                          shape=spec.shape,
                          dtype=spec.dtype)
    else:
        raise NotImplementedError

class DMCEnv(core.Env):
    def __init__(self,
                 domain_name: str,
                 task_name: str,
                 task_kwargs: Optional[Dict] = {},
                 environment_kwargs=None):
        assert 'random' in task_kwargs, 'please specify a seed, for deterministic behaviour'

        self._env = suite.load(domain_name=domain_name,
                               task_name=task_name,
                               task_kwargs=task_kwargs,
                               environment_kwargs=environment_kwargs)
        self.action_space = dmc_spec2gym_space(self._env.action_spec())

        self.observation_space = dmc_spec2gym_space(
            self._env.observation_spec())

        self.seed(seed=task_kwargs['random'])

    def __getattr__(self, name):
        return getattr(self._env, name)

    def step(self, action):
        assert self.action_space.contains(action)

        time_step = self._env.step(action)
        reward = time_step.reward or 0
        done = time_step.last()
        obs = time_step.observation

        info = {}
        if done and time_step.discount == 1.0:
            info['TimeLimit.truncated'] = True

        return obs, reward, done, info

    def reset(self):
        time_step = self._env.reset()
        return time_step.observation

def make_env(env, seed):
    domain_name, task_name = env.split('-')
    env = DMCEnv(
        domain_name = domain_name,
        task_name   = task_name,
        task_kwargs = {'random': seed}
    )
    if isinstance(env.observation_space, gym.spaces.Dict):
        env = gym.wrappers.FlattenObservation(env)
    return env

env = make_env('humanoid-run', 42)
for t in range(10000):
    action = env.action_space.sample()
    next_state, reward, done, info = env.step(action)
```

kevinzakka, 2022-07-08:
OOC, how big is the impact?

dyth, 2022-07-08:
I think nearly up to 1.5x

nimrod-gileadi, 2022-07-14:
Just so I understand, this script does nothing with JAX other than importing it, right? dm_control itself doesn't do anything with JAX.

If it's easy, could you please do the timing around the environment loop rather than timing the whole script?

My first hypothesis is that it's to do with the time taken to import the libraries rather than runtime speed, and that there may be some caching which makes whichever script you run second load faster.

twni2016, 2022-07-14:
Thanks for your script! This makes sense.

nimrod-gileadi, 2022-07-18:
We still don't know the cause for this.

It's not a high priority issue for us, so it's unlikely to be fixed soon.
For now, could you import in alphabetical order? 😝

saran-t, 2022-08-04:
@dyth could you please check if the issue still persists as of https://github.com/deepmind/dm_control/commit/ac6d2cd7af7f6d20bbc0e51df8ba41016a07f1f9 ?

twni2016, 2022-08-04:
@saran-t I can confirm that dm_control v1.0.5 solved the issue! The importing order does not matter now. Moreover, it is ~20% faster than before (`fast=True` in v1.0.3). Thank you!

closed 2022-08-05 by dyth

saran-t, 2022-08-05:
That's great to hear! I still have no idea how the Jax import order enters into the picture though. The fix is to do with `dm_control` performance generally.