google-deepmind/dm_control #321: Mujoco 2.0.0 with dm_control

oofmeister27, 2022-07-12:
Hello,

I am working with another repository that needs Mujoco version 2.0.0 installed for it to work (as well as dm_control). I downloaded Mujoco 2.0.0 as well as the license from this [website](http://roboti.us/download.html) and put it in the .mujoco folder. I tested out mujoco out and it worked as expected.

However, when I install dm_control it automatically installs mujoco 2.2.0. How can I download Mujoco 2.0.0 with the dm_control library? I did install mujoco-py version 2.0.2.8 but for some reason, mujoco 2.0.0 isn't being detected.  Any help would be appreciated.

saran-t, 2022-08-04:
If you need MuJoCo 2.0.0, the easiest way is to `pip install dm_control==0.0.364896371` since that's the last version on PyPI that supports that version of MuJoCo.

I can't help with `mujoco-py` I'm afraid, but the same comment goes there too: you need to figure out what the latest version of `mujoco-py` that supports MuJoCo 2.0.0.

Any reason why you have to use 2.0.0?

oofmeister27, 2022-08-04:
I got it to work. I didn't have to run any pip command to install mujoco. I just downloaded Mujoco 2.0.0 from the Roboti website and added a license key and it ended up working. Thanks for the help!

closed 2022-08-04 by oofmeister27