google-deepmind/dm_control #328: How to know the physical meaning of each dimension of state space?

zdhNarsil, 2022-07-28:
I would like to write a wrapper for dm control environment and zero out some specific dimensions of the state. But I don't know the detailed physical meaning (i.e., position / velocity) of each individual dimension for each environment. I wonder where could I find these information?

kevinzakka, 2022-07-29:
Right now, you're gonna have to look at the source code for each environment. I don't think it's particularly hard to parse but I do agree that it would be nice to have comments or a white-paper to describe each in detail :)

zdhNarsil, 2022-07-29:
@kevinzakka Hi Kevin, thanks for your reply! I have no idea where to look at in the source code (i.e., I cannot find the code corresponding to some specific environment), would you mind share any pointer as an example?

kevinzakka, 2022-07-30:
What environments are you looking at? It will differ for the different suites in this repo.

* For the dm_control `suite`, look at the `*.py` files defining each environment and check the observation dict. For example, [`acrobot.py`](https://github.com/deepmind/dm_control/blob/main/dm_control/suite/acrobot.py#L108)
* For the dm_control `manipulation` suite and `locomotion` suite, these are composer tasks so look at the `__init__` function in each task definition and check what entity observable and task observables are defined. Usually the observations are returned as a dict so it's probably easier if you load the environment, step it and inspect the observation dict. But if you want to read code to learn more, look at the `composer.Task` definition and whatever `composer.Entity` is being used in the task.

Happy to answer more questions if you have any.

zdhNarsil, 2022-07-30:
Yes, I mean dm_control `suite`. 

In the `*.py` files you mentioned, there is a `self.get_observation()` function that returns a dict `obs`. But how does dm_control transform this python dict into a numpy array? I found this [`flatter_observation()`](https://github.com/deepmind/dm_control/blob/20cef21e25/dm_control/rl/control.py#L367) but I am not sure.

closed 2022-08-25 by zdhNarsil