google-deepmind/dm_control #331: Wrong angles

Melanol, 2022-08-04:
This is now my model looks in simulate:
[image]

This is how it looks when I import it with `mjcf.from_path("tripod.xml")` into the tutorial task:
[image]

yuvaltassa, 2022-08-04:
This would be easier to diagnose if you included the actual XML. Off the top of my head I suspect the base arena in our `mjcf` tutorial sets `<compiler angle="radian"/>` somewhere, while your original uses degrees?

Melanol, 2022-08-04:
@yuvaltassa I do use degrees in my .xml. I think you are right, I did
```
    model = mjcf.from_path("tripod.xml")
    model.compiler.angle = "degree"

    return model
```

and received

`ValueError: Conflicting values for attribute `angle`: radian vs degree`

So, what is the fix?

Melanol, 2022-08-04:
OK, I manually changed all degrees in my .xml to radians, and it works.

closed 2022-08-04 by Melanol