google-deepmind/dm_control #338: cannot open .stl mesh files

NgChucky, 2022-09-03:
I tried to load this xml file in simulate.exe:
(my models are stored in the MuJoCo models directory, including the stl file)
```
<mujoco> 
<compiler angle="radian" coordinate="local" inertiafromgeom="true" settotalmass="25" meshdir="C:\Users\Nagaraju Chukkala\Documents\programs\MuJoCo models\"/>  
    <worldbody>  
        <mesh name='monkey' file="untitled.stl" />  
    </worldbody>  
</mujoco>
```
It says "unrecognized element mesh line:4"
Where is the documentation for mjcf and dm_control? How do I get to understand it?

yuvaltassa, 2022-09-03:
The mesh needs to referenced by a geom.

[Here](https://mujoco.readthedocs.io/en/latest/XMLreference.html) is the documentation.

closed 2022-09-03 by yuvaltassa

NgChucky, 2022-09-03:
Yuval Tassa! That's a privilege. I came here to close this issue as I had figured it, and saw Yuval's reply.

commit 2e2cf9d7 (2022-08-03): Compute moment of inertia for concave and boundary meshes. Resolve #338.

- Mesh inertias can now be computed exactly for well-formed (no holes) non-convex meshes.
- To activate this feature, set `<compiler exactmeshinertia="true">` (defaults to `false`). This default may change in the future.
- Added `<geom shellinertia="true/false">` (defaults to `false`). When true, geom inertia is computed assuming all the mass is concentrated on the surface, and `density` is interpreted as surface density (mass/area). Currently only mesh geoms are supported.

PiperOrigin-RevId: 464368395
Change-Id: I02fd6 [... cut]