google-deepmind/dm_control #341: help

NgChucky, 2022-09-04:
I get this error in simultate.exe when I run the following: "mass and inertia of moving bodies must be larger than mjMINVAL"
```
<mujoco>
    <size nconmax="6000" njmax="6000" nstack="50000000"/>
    <default>
        <default class='x'>
            <joint type="hinge" damping=".2" stiffness="1" armature=".01" limited="true" solimplimit="0 .99 .01"/>
            <default class='y'>
                <geom type='sphere' mass='1' condim="1" friction=".7" solimp=".9 .99 .003" solref=".015 1"/>
            </default>
        </default>
    </default>
    <worldbody>
        <light name='top_light' pos='1 5 20'/>
        <light name='t_light' pos='5 1 10'/>
        <geom name='floor' type='plane' size='0 0 0.5' condim='3'/>
        <body>
        <freejoint/>
        <geom name='floor_box' size='1 1 1' type='box' pos='-1 0 1.5'/>
        </body>
        <body>
            <freejoint name='root'/>
            <body>
                <joint axis="0 1 0" range="-45 45" class="x"/>
                <geom name='link1' mass="100" size='0.2' class='y' pos='-4 0 3'/>
                <body>
                    <joint axis="0 1 0" range="-45 45" class="x"/>
                    <geom name='link2' mass="50"size='0.2' class='y' pos='-3 0 3'/>
                    <body>
                        <joint axis="0 1 0" range="-45 45" class="x"/>
                        <geom name='link3' mass="25" size='0.2' class='y' pos='-2 0 3'/>
                        <body>
                            <joint axis="0 1 0" range="-45 45" class="x"/>
                            <geom name='link4' mass="12" size='0.2' class='y' pos='-1 0 3'/>
                            <body>
                                <joint axis="0 1 0" range="-45 45" class="x"/>
                                <geom name='link5' mass="6" size='0.2' class='y' pos='0 0 3'/>
                                <body>
                                    <joint axis="0 1 0" range="-45 45" class="x"/>
                                    <geom name='link6' mass="3" size='0.2' class='y' pos='1 0 3'/>
                                    <body>
                                        <joint axis="0 1 0" range="-45 45" class="x"/>
                                        <geom name='link7' mass="1" size='0.2' class='y' pos='2 0 3'/>
                                    </body>
                                </body>
                            </body>
                        </body>
                    </body>
                </body>
            </body>
        </body>
    </worldbody>
</mujoco>
```

nimrod-gileadi, 2022-09-05:
The issue is that the second `freejoint` is defined in a body that has no geom (and thus no mass).

You can get around that by removing the massless wrapper body, and just putting the `freejoint` tag in the first bead in the chain (replacing the joint you have there already), so that the connection between `worldbody` and the bead is a free joint.

NgChucky, 2022-09-05:
Thanks, Nimrod and Saran.

closed 2022-09-05 by NgChucky