google-deepmind/dm_control #343: Advice modeling simple block sliding on floor

bryanvis, 2022-09-07:
I'm getting started with dm_control and am trying to model a bare bones environment: a box sliding around a floor.
To do this, I adapted from [mujoco-worldgen](https://github.com/openai/mujoco-worldgen) just a box with 3 slide joints and corresponding actuators. I also use the floor arena and then edited down locomotion walker/target tasks to be as simple as possible.
XML here. Note that if I don't include joint armature = 1, then it immediately fails w/ `Nan, Inf or huge value in QACC at DOF 5` at time t=0.000. When I include it though, the box does not move.
```
<mujoco>  
	<compiler angle="radian"/>
	<default>
    <joint armature="1" />
	</default>
	<worldbody>
		<body name="agent" pos="0.2 0.2 0.4">
      		<camera name="fake_egocentric"  pos=".25 0 .11" xyaxes="0 -1 0 0 0 1" fovy="90" />
			<joint axis="1 0 0" damping="10" name="tx" pos="0 0 0" type="slide"></joint>
			<joint axis="0 1 0" damping="10" name="ty" pos="0 0 0" type="slide"></joint>
			<joint axis="0 0 1" damping="10000" name="tz" pos="0 0 0" type="slide"></joint>
			<joint axis="0 0 1" damping="10"  name="rz" pos="0 0 0" type="hinge"></joint>
			<geom name="agent"  mass="1" pos="0 0 0" rgba="0.52734375 0.8046875 0.9765625 1" size="0.2 0.2 0.2" type="box" euler="1.57 0 0"></geom>
		</body>
	</worldbody>
	<actuator>
		<motor name="motortx" gear="100" joint="tx" ></motor>
		<motor name="motorty" gear="100" joint="ty"></motor>
		<motor name="motorrz" gear="100" joint="rz"></motor>
	</actuator>
</mujoco>
```
Full zip of code is attached. 

[ice 2.zip](https://github.com/deepmind/dm_control/files/9504813/ice.2.zip)

yuvaltassa, 2022-09-07:
The model you posted loads with no issue and has only 4 DoFs (yet the error is complaining about DOF 5), so the problem is elsewhere, something about the walker that you are adding?

bryanvis, 2022-09-07:
Thanks Yuval, the runnable code is in the zip above but walker is here below:

```
import os

from dm_control import composer
from dm_control import mjcf
from dm_control.composer.observation import observable
from dm_control.locomotion.walkers import base
from dm_control.locomotion.walkers import legacy_base
import numpy as np

_XML_DIRNAME = os.path.join(os.path.dirname(__file__), './')
_XML_FILENAME = 'ice.xml'

class Ice(legacy_base.Walker):

  def _build(self, name='walker', initializer=None):
    """Build a walker.

    Args:
      name: name of the walker.
      initializer: (Optional) A `WalkerInitializer` object.
    """
    super()._build(initializer=initializer)
    self._mjcf_root = mjcf.from_path(os.path.join(_XML_DIRNAME, _XML_FILENAME))
    if name:
      self._mjcf_root.model = name

  def initialize_episode(self, physics, random_state):
    pass

  def apply_action(self, physics, action, random_state):
    super().apply_action(physics, action, random_state)

  def _build_observables(self):
    return IceObservables(self)

  @property
  def mjcf_model(self):
    return self._mjcf_root

  @property
  def upright_pose(self):
    return base.WalkerPose()

  @composer.cached_property
  def actuators(self):
    return self._mjcf_root.find_all('actuator')

  # Used by parent class.
  @composer.cached_property
  def root_body(self):
    return self._mjcf_root.find('body', 'agent')

  @composer.cached_property
  def bodies(self):
    return tuple(self.mjcf_model.find_all('body'))

  @composer.cached_property
  def observable_joints(self):
    return [actuator.joint for actuator in self.actuators] 

  @composer.cached_property
  def egocentric_camera(self):
    return self._mjcf_root.find('camera', 'fake_egocentric')

  @property
  def end_effectors(self):
    return ()

  @composer.cached_property
  def ground_contact_geoms(self):
    return ()

class IceObservables(legacy_base.WalkerObservables):
  @property
  def proprioception(self):
    return [self.position, self.orientation, self.joints_vel]
```
I'm stumped as to why it's saying 5 dof when no other joints are added? but I'm sure I overlooked something trivial 😄

bryanvis, 2022-09-07:
Ah, in the target task this line `self._walker.create_root_joints(self._arena.attach(self._walker))` is adding a joint

bryanvis, 2022-09-08:
Revisited w/ the composer tutorial. I compressed it down to a single file:
```python
from dm_control import composer
from dm_control.composer.observation import observable
import numpy as np
from dm_control.locomotion.arenas import floors
from dm_control import mjcf

def make_frank():
  random_state = np.random.RandomState(42)
  rgba = random_state.uniform([0, 0, 0, 1], [1, 1, 1, 1])
  model = mjcf.RootElement()
  model.compiler.angle = 'radian'  # Use radians.
  box = model.worldbody.add('body', pos=[0.5,0.5,0.5])
  box.add(
      'geom', name='torso', type='box', size=[0.5,0.5,0.5], rgba=rgba)
  x = box.add("joint", axis=[1,0,0], type="slide", pos=(0,5,0))
  model.actuator.add("motor", gear="100", joint=x)
  # y = box.add("joint", axis=[0,1,0], type="slide", name="my", pos=(0,0,0))
  # model.actuator.add("motor", gear="100", joint=y)
  return model

class Creature(composer.Entity):
  def _build(self):
    self._model = make_frank()
  def _build_observables(self):
    return CreatureObservables(self)
  @property
  def mjcf_model(self):
    return self._model
  @property
  def actuators(self):
    return tuple(self._model.find_all('actuator'))

class CreatureObservables(composer.Observables):
  @composer.observable
  def joint_positions(self):
    all_joints = self._entity.mjcf_model.find_all('joint')
    return observable.MJCFFeature('qpos', all_joints)
  @composer.observable
  def joint_velocities(self):
    all_joints = self._entity.mjcf_model.find_all('joint')
    return observable.MJCFFeature('qvel', all_joints)

NUM_SUBSTEPS = 25
class BasicTask(composer.Task):
  def __init__(self, creature):
    self._creature = creature
    self._arena = floors.Floor()
    self._arena.add_free_entity(self._creature)
    self._arena.mjcf_model.worldbody.add('light', pos=(0, 0, 4))
    self._creature.observables.joint_positions.enabled = True
    self._creature.observables.joint_velocities.enabled = True
    self.control_timestep = NUM_SUBSTEPS * self.physics_timestep
  @property
  def root_entity(self):
    return self._arena
  def initialize_episode_mjcf(self, random_state):
    pass
  def initialize_episode(self, physics, random_state):
    pass
  def get_reward(self, physics):
    return 0

def get_task():
  creature = Creature()
  task = BasicTask(creature)
  env = composer.Environment(task, random_state=np.random.RandomState(42))
  return env
```
Some observations:
1. this runs but the cube "stutters" at the beginning and wobbles eventually. Adding damping makes it stay still.
2. adding the Y joint/motor immediately causes the simulation to break, with the same error but DOF 0. 
3. substituting `make_frank()` with `make_creature()` as in the tutorial runs fine.

bryanvis, 2022-09-08:
To try to debug further I got the xml at runtime:

```xml
<mujoco model="floor">
  <compiler boundmass="1e-05" boundinertia="1e-11" coordinate="local" angle="radian" eulerseq="xyz"/>
  <option timestep="0.002" integrator="RK4" cone="elliptic" noslip_iterations="5" noslip_tolerance="0.0"/>
  <visual>
    <headlight ambient="0.40000000000000002 0.40000000000000002 0.40000000000000002" diffuse="0.80000000000000004 0.80000000000000004 0.80000000000000004" specular="0.10000000000000001 0.10000000000000001 0.10000000000000001"/>
    <map znear="0.01"/>
    <scale forcewidth="0.01" contactwidth="0.06" contactheight="0.01" jointwidth="0.01" framelength="0.3" framewidth="0.01"/>
  </visual>
  <default>
    <default class="/"/>
    <default class="unnamed_model/"/>
  </default>
  <asset>
    <texture name="groundplane" type="2d" builtin="checker" rgb1="0.20000000000000001 0.29999999999999999 0.40000000000000002" rgb2="0.10000000000000001 0.20000000000000001 0.29999999999999999" mark="edge" markrgb="0.80000000000000004 0.80000000000000004 0.80000000000000004" width="200" height="200"/>
    <material name="groundplane" class="/" texture="groundplane" texrepeat="2 2" texuniform="true" reflectance="0.2"/>
  </asset>
  <worldbody>
    <geom name="groundplane" class="/" type="plane" size="8 8 0.25" material="groundplane"/>
    <camera name="top_camera" class="/" fovy="10.058147163570894" pos="0 0 100" quat="1 0 0 0"/>
    <body name="unnamed_model/">
      <freejoint name="unnamed_model/"/>
      <body name="unnamed_model//unnamed_body_0" pos="0.050000000000000003 0.050000000000000003 0.050000000000000003">
        <geom name="unnamed_model/torso" class="unnamed_model/" type="box" size="0.050000000000000003 0.050000000000000003 0.050000000000000003" rgba="0.15601864044243652 0.15599452033620265 0.058083612168199461 1" mass="1.0"/>
        <joint name="unnamed_model//unnamed_joint_0" class="unnamed_model/" type="slide" pos="0 0 0" axis="1 0 0"/>
        <joint name="unnamed_model/my" class="unnamed_model/" type="slide" pos="0 0 0" axis="0 1 0"/>
      </body>
    </body>
    <light name="//unnamed_light_0" class="/" pos="0 0 4"/>
  </worldbody>
  <actuator>
    <motor name="unnamed_model//unnamed_actuator_0" class="unnamed_model/" gear="100" joint="unnamed_model//unnamed_joint_0"/>
    <motor name="unnamed_model//unnamed_actuator_1" class="unnamed_model/" gear="100" joint="unnamed_model/my"/>
  </actuator>
</mujoco>
```
Some things I tried: changing the integrator to RK4; reducing the timestep; changing solver params.

yuvaltassa, 2022-09-09:
This is a very strange model, not clear to me what you are trying to do. Maybe if I understood your goal I could help you.

You have a body with no mass that only compiles because of the `boundmass` and `boundinertia` (so really it has a very tiny mass), and then this body has a chid with two slider joints? I don't understand why you would do that. Do you want the body to have 6 DoFs or 2? These chained 7 DoFs make no sense. Also your actuators are far too strong.

bryanvis, 2022-09-09:
@yuvaltassa Sorry, that XML is the compiled one at runtime, maybe the code in the comment before that makes more sense.
The goal is to have a cube that is the entire agent; it should slide along the floor, and be able to move in the x and y direction as well as rotate on the xy plane.
I've been able to accomplish the x/y direction but not the rotation — I'll try to get a minimal working code for that in a followup.

yuvaltassa, 2022-09-09:
So you want the cube to have 4 DoFs? How could it rotate if it can't move up? The corners will collide with the floor.

bryanvis, 2022-09-09:
By "rotate on the xy plane", I just meant that the cube should be able to spin as you might spin a top — don't need to move up. Does that make sense?

bryanvis, 2022-09-09:
XML:
```xml
<mujoco model="ice">  
	<compiler angle="radian"/>
	<default class="root">
	      <joint pos='0 0 0' limited='false' armature='0' damping='0' stiffness='0'/>
	</default>
	<worldbody>
		<body name="agent" pos="0.05 0.05 0.05">
      		<camera name="fake_egocentric"  pos="5 0 5" xyaxes="0 -1 0 0 0 1" fovy="90" />
			<joint class="root" axis="1 0 0"  name="tx" pos="0 0 0" type="slide"></joint>
			<joint class="root" axis="0 1 0"  name="ty" pos="0 0 0" type="slide"></joint>
			<geom  mass="1" pos="0 0 0" rgba="0.52734375 0.8046875 0.9765625 1" size="0.05 0.05 0.05" type="box" euler="1.57 0 0"></geom>
		</body>
	</worldbody>
	<actuator>
		<motor name="motortx" gear="100" joint="tx" ></motor>
		<motor name="motorty" gear="100" joint="ty"></motor>
	</actuator>
</mujoco>
```
Code:
```python
from dm_control import composer
from dm_control.composer.observation import observable
import numpy as np
from dm_control.locomotion.soccer.pitch import Pitch

from dm_control import mjcf
import os
_XML_DIRNAME = os.path.join(os.path.dirname(__file__), '../ice/')
_XML_FILENAME = 'ice.xml'
random_state = np.random.RandomState(42)

class Creature(composer.Entity):
  """A multi-legged creature derived from `composer.Entity`."""
  def _build(self):
    self._model = mjcf.from_path(os.path.join(_XML_DIRNAME, _XML_FILENAME))
  def _build_observables(self):
    return CreatureObservables(self)
  @property
  def mjcf_model(self):
    return self._model
  @property
  def actuators(self):
    return tuple(self._model.find_all('actuator'))
  def create_root_joints(self, attachment_frame):
    root_class = self._model.find('default', 'root')
    root_x = attachment_frame.add(
        'joint', name='root_x', type='slide', axis=[1, 0, 0], dclass=root_class)
    root_y = attachment_frame.add(
        'joint', name='root_y', type='slide', axis=[0, 1, 0], dclass=root_class)

# Add simple observable features for joint angles and velocities.
class CreatureObservables(composer.Observables):
  @composer.observable
  def joint_positions(self):
    all_joints = self._entity.mjcf_model.find_all('joint')
    return observable.MJCFFeature('qpos', all_joints)
  @composer.observable
  def joint_velocities(self):
    all_joints = self._entity.mjcf_model.find_all('joint')
    return observable.MJCFFeature('qvel', all_joints)

from dm_control.locomotion.arenas import floors
NUM_SUBSTEPS = 25
class BasicTask(composer.Task):
  def __init__(self, creature):
    self._creature = creature
    self._arena = floors.Floor()
    self._creature.create_root_joints(self._arena.attach(self._creature))
    self._arena.mjcf_model.worldbody.add('light', pos=(0, 0, 20))
    self._creature.observables.joint_positions.enabled = True
    self._creature.observables.joint_velocities.enabled = True
    self.control_timestep = NUM_SUBSTEPS * self.physics_timestep
  @property
  def root_entity(self):
    return self._arena
  def initialize_episode_mjcf(self, random_state):
    pass
  def initialize_episode(self, physics, random_state):
    pass
  def get_reward(self, physics):
    return 0

def get_task():
  creature = Creature()
  task = BasicTask(creature)
  env = composer.Environment(task, random_state=np.random.RandomState(42))
  return env
```
Here's what I got working for x/y direction. Admittedly, I pieced this together using the soccer example, and it'd be great to have guidance on the right way to model this.

yuvaltassa, 2022-09-09:
From your description, sounds like you want to add a **hinge** joint with an axis in the z-direction?

Not sure what you mean about `euler`, in what way does it "not work"?

bryanvis, 2022-09-09:
@yuvaltassa I tried adding such a hinge as so:
```xml
<mujoco model="ice">  
	<compiler angle="radian"/>
	<default class="root">
	      <joint pos='0 0 0' limited='false' armature='0' damping='0' stiffness='0'/>
	</default>
	<worldbody>
		<body name="agent" pos="0 0 0.05">
      		<camera name="fake_egocentric"  pos="5 0 5" xyaxes="0 -1 0 0 0 1" fovy="90" />
			<joint class="root" axis="1 0 0"  name="tx" pos="0 0 0" type="slide"></joint>
			<joint class="root" axis="0 1 0"  name="ty" pos="0 0 0" type="slide"></joint>
			<joint class="root" axis="0 0 1"  name="rz" pos="0 0 0" type="hinge"></joint>
			<geom  mass="1" pos="0 0 0" rgba="0.52734375 0.8046875 0.9765625 1" size="0.05 0.05 0.05" type="box" euler="1.57 0 0"></geom>
		</body>
	</worldbody>
	<actuator>
		<motor name="motortx" gear="100" joint="tx" ></motor>
		<motor name="motorty" gear="100" joint="ty"></motor>
		<motor name="motorrz" gear="10" joint="rz"></motor> 
	</actuator>
</mujoco>
```
(I figured out the create_root_joints was completely unnecessary)

But here's a video when applying [0,0,1] forces. I expect it to spin around the center of the cube.
https://user-images.githubusercontent.com/27381586/189420556-1cb75def-6f5b-47c1-85bf-935108207026.mov

And here's a video applying [1,0,0] force. It works when the joint/motor are removed.
https://user-images.githubusercontent.com/27381586/189420806-2bfc59c7-7d11-4a06-b92a-ce785b8db277.mov

bryanvis, 2022-09-10:
Ok, I think this was simply due to friction and setting "condim" to 1 for both the floor & the cube gets me the behavior I was looking for! (If this is a poor solution do let me know though :) )