google-deepmind/dm_control #347: Find the real coordinate of the object

neuronphysics, 2022-09-07:
Hi
I am trying to get the coordinates of different agent in the `dm_control ` environment. So far this is my attempt 
```
physics = mujoco.Physics.from_xml_string(*get_model_and_assets("Humanoid-v2"))
with physics.reset_context():
    """Returns a copy of the generalized positions (system configuration)."""
    position=physics.data.qpos[:]
    print(position.shape)
    """Returns a copy of the generalized velocities."""
    velocity=physics.data.qvel[:]
    print(velocity.shape)
    observation = np.concatenate([position, velocity]).ravel()
    
scene_option = mujoco.wrapper.core.MjvOption()
scene_option.flags[enums.mjtVisFlag.mjVIS_JOINT] = True
pixels = physics.render(scene_option=scene_option, camera_id=1, segmentation=True)
fig, ax = plt.subplots(1, 1)
ax.imshow(pixels[:,:,0])

ax.set_axis_off()
"""Observations consist of an OrderedDict containing one or more NumPy arrays"""
#PIL.Image.fromarray(pixels)
def xyz2pixels(xyz, camera_matrix):
    """ Project 3D locations to pixel locations using the camera matrix """ 
    #https://github.com/rinuboney/FPAC/blob/b918755d9137fcfb77f2a855db1d3c661444bdf0/utils.py
    xyzs = np.ones((xyz.shape[0], xyz.shape[1]+1))
    xyzs[:, :xyz.shape[1]] = xyz
    xs, ys, s = camera_matrix.dot(xyzs.T)
    return xs/s, ys/s

def gt_keypoints():
    """ Extract 2D pixel locations of objects in the environment """
    camera_matrix = mujoco.Camera(physics, height=84, width=84, camera_id=1).matrix
    xyz = physics.named.data.geom_xpos.copy()
    box_pos = physics.named.data.geom_xpos['head']
    box_mat = physics.named.data.geom_xmat['head'].reshape(3, 3)
    box_size = physics.named.model.geom_size['head']
    offsets = np.array([-1, 1]) * box_size[:, None]
    xyz_local = np.stack(itertools.product(*offsets)).T
    print(f"local:{xyz_local}")
    xyz_global = box_pos[:, None] + box_mat @ xyz_local  
    print(f"global:{xyz_global}")  
    # # Camera matrices multiply homogenous [x, y, z, 1] vectors.
    corners_homogeneous = np.ones((4, xyz_global.shape[1]), dtype=float)
    corners_homogeneous[:3, :] = xyz_global
    u, v, z = camera_matrix @ corners_homogeneous
    # x and y are in the pixel coordinate system of the head.
    ax.plot(u/z,v/z,'m*')
    x = u / z
    y = v / z
    return xyz2pixels(xyz,camera_matrix)

x,y=gt_keypoints()
# Render the camera view and overlay the projected corner coordinates.

ax.plot(x,y,'r+')
plt.show()
```
There is an offset between the x ,y points and the [real image](https://i.postimg.cc/d008TwFD/Screen-Shot-2022-09-07-at-1-22-11-PM.png). Would anyone clarify how I can map these points exactly on the object?

yuvaltassa, 2022-09-11:
You're following the example in the colab, yes?

neuronphysics, 2022-09-11:
yes, I did.

yuvaltassa, 2022-09-11:
Is it not just that the resolution is different? 
This is using default res
`pixels = physics.render(scene_option=scene_option, camera_id=1, segmentation=True)`
This is using 84x84
`camera_matrix = mujoco.Camera(physics, height=84, width=84, camera_id=1).matrix`

closed 2022-09-11 by yuvaltassa

neuronphysics, 2022-09-11:
How can I fix it?

yuvaltassa, 2022-09-11:
Ummm maybe
`pixels = physics.render(scene_option=scene_option, camera_id=1, segmentation=True, height=84, width=84)`