google-deepmind/dm_control #349: Regarding the setting of sphere parameters

miyukin73, 2022-09-27:
Hello, 

I have a question regarding the setting of sphere parameters.

I wanted to use Mujoco to simulate the pouring of marbles from a cup as shown in the first movie, 
and I have created the simulation shown below with a lot of help....!

https://user-images.githubusercontent.com/63595993/192478534-9feb97ac-95ee-47cb-9a39-6e3216c84c70.mp4

https://user-images.githubusercontent.com/63595993/192479910-049820dd-bd61-44f0-a736-1cec58d316cb.mov

Thanks to your many instructions, I was able to complete a rough simulation, but I think I still need to make minor tweaks.

Specifically, the problem is the difference that the simulation shows a **smooth** outflow from the **lower part** of the marble, while in reality the structure of the marble **collapses** and outflow occurs from the **upper** part of the marble.

I thought that the **friction coefficients** of the marbles and cups had something to do with this fine-tuning.
So I have changed the values of friction in geom, but there is no improvement.

I know this is a very detailed question, but I would like to know if there are other parameters that might be involved...!

Also, I read on [this site](https://mujoco.readthedocs.io/en/latest/XMLreference.html) that the values of sliding, torsional, and rolling friction can be changed friction of "geom", but is it possible to set the **static friction coefficient**?

Furthermore, I would like to adjust the value of the **surface hardness** of the marble, which seems to be related to it.

If there are any other values that I can change by myself, I would appreciate it if you could let me know.

Thank you in advance.

yuvaltassa, 2022-09-28:
There is a bigger issue here. The balls are just "particles" with 3 slides DoFs.
What you really want is 6DoF balls that can roll. Until you do that adding friction will give you strange behaviour, as the balls cannot rotate.

Really this should be an option of the `particle` composite, and will be easy to implement.
I'll make an internal feature request for this. If you like you can open a new feature-request issue on GitHub which I will leave open until it is implemented.

closed 2022-09-28 by yuvaltassa

miyukin73, 2022-09-28:
Hello, @yuvaltassa 

Thank you very much for taking the time to reply.

I am not sure if I am replying to the right question as this issue is currently closed, but I would like to confirm two points.

(1) I have currently defined type="sphere" and set freejoint as shown in the image below, but I am still not able to rotate it?
I did it so because I was not sure if it is possible to specify the position of each individual particle when it is defined as particles or not.
[image]

(2) Thank you for considering it as an option for Particle composite. However, I did not understand your response to the message,
> If you like you can open a new feature-request issue on GitHub which I will leave open until it is implemented.

Is it sufficient if I just tell you that I would like you to consider it?

Thank you in advance.

yuvaltassa, 2022-09-28:
Ah hold on, how did you make that `<freejoint>`?

Yes that is exactly what I meant, but did you do it by saving an XML with a composite and replacing the sliders with a `freejoint` or did you not use the composite at all and just build your model from scratch?

In the case that you already have freejoints then your question totally makes sense.

Please attach the model that you are using and we will continue the discussion from there.

miyukin73, 2022-09-29:
Hi, @yuvaltassa 

Thank you for contacting me.

I don't use "composite"
because I thought that composite cannot specify the position of each individual particle.
However, if it is possible, I'd like to know how to do it.

I just set up freejoint based on [this site](https://mujoco.readthedocs.io/en/latest/modeling.html).

[image]

 
I attached is the XML file.
[particle.zip](https://github.com/deepmind/dm_control/files/9669832/particle.zip)

I am afraid I am a beginner and don't understand a lot of points, but I would appreciate it if you could check it for me.

Thank you very much in advance.

yuvaltassa, 2022-10-18:
Hi,

Just tried to take a look. Your model is not loadable. It says `<include file="Cup_rotate_0722.xml"/>`, but you did not include this file or its dependents. Please attach a loadable model and I will look again.

miyukin73, 2022-10-19:
Hi, thank you for checking the file and I'm sorry for the error.

I attached my file again.
[particle.zip](https://github.com/deepmind/dm_control/files/9817098/particle.zip)

Thank you.