google-deepmind/dm_control #350: mujoco.FatalError in Ubuntu 18.04.5

hydro-man, 2022-10-17:
when i run memory_maze i met a Error below:
`/home/DQC/anaconda3/envs/test/lib/python3.7/site-packages/glfw/__init__.py:912: GLFWError: (65544) b'X11: The DISPLAY environment variable is missing'
  warnings.warn(message, GLFWError)
/home/DQC/anaconda3/envs/test/lib/python3.7/site-packages/glfw/__init__.py:912: GLFWError: (65537) b'The GLFW library is not initialized'
  warnings.warn(message, GLFWError)
Traceback (most recent call last):
  File "/home/DQC/demo/test.py", line 4, in <module>
    env = gym.make('dmc_memory_maze:MemoryMaze-9x9-v0')
  File "/home/DQC/anaconda3/envs/test/lib/python3.7/site-packages/gym/envs/registration.py", line 640, in make
    env = env_creator(**_kwargs)
  File "/home/DQC/anaconda3/envs/test/lib/python3.7/site-packages/dmc_memory_maze-0.2.0-py3.7.egg/dmc_memory_maze/__init__.py", line 17, in _make_gym_env
    dmenv = dm_task(**kwargs)
  File "/home/DQC/anaconda3/envs/test/lib/python3.7/site-packages/dmc_memory_maze-0.2.0-py3.7.egg/dmc_memory_maze/tasks.py", line 25, in memory_maze_9x9
    return _memory_maze(9, 3, 250, **kwargs)
  File "/home/DQC/anaconda3/envs/test/lib/python3.7/site-packages/dmc_memory_maze-0.2.0-py3.7.egg/dmc_memory_maze/tasks.py", line 105, in _memory_maze
    strip_singleton_obs_buffer_dim=True)
  File "/home/DQC/anaconda3/envs/test/lib/python3.7/site-packages/dm_control/composer/environment.py", line 331, in __init__
    delayed_observation_padding=delayed_observation_padding)
  File "/home/DQC/anaconda3/envs/test/lib/python3.7/site-packages/dm_control/composer/environment.py", line 221, in __init__
    self._recompile_physics_and_update_observables()
  File "/home/DQC/anaconda3/envs/test/lib/python3.7/site-packages/dm_control/composer/environment.py", line 243, in _recompile_physics_and_update_observables
    self._observation_updater.reset(self._physics_proxy, self._random_state)
  File "/home/DQC/anaconda3/envs/test/lib/python3.7/site-packages/dm_control/composer/observation/updater.py", line 165, in reset
    enabled.observation_callable())
  File "/home/DQC/anaconda3/envs/test/lib/python3.7/site-packages/dm_control/composer/observation/observable/mjcf.py", line 273, in get_observation
    render_flag_overrides=self._render_flag_overrides)
  File "/home/DQC/anaconda3/envs/test/lib/python3.7/site-packages/dm_control/mujoco/engine.py", line 220, in render
    scene_callback=scene_callback)
  File "/home/DQC/anaconda3/envs/test/lib/python3.7/site-packages/dm_control/mujoco/engine.py", line 694, in __init__
    if self._physics.contexts.mujoco is not None:
  File "/home/DQC/anaconda3/envs/test/lib/python3.7/site-packages/dm_control/mujoco/engine.py", line 525, in contexts
    self._make_rendering_contexts()
  File "/home/DQC/anaconda3/envs/test/lib/python3.7/site-packages/dm_control/mujoco/engine.py", line 511, in _make_rendering_contexts
    mujoco_context = wrapper.MjrContext(self.model, render_context)
  File "/home/DQC/anaconda3/envs/test/lib/python3.7/site-packages/dm_control/mujoco/wrapper/core.py", line 606, in __init__
    ptr = ctx.call(mujoco.MjrContext, model.ptr, font_scale)
  File "/home/DQC/anaconda3/envs/test/lib/python3.7/site-packages/dm_control/_render/executor/render_executor.py", line 136, in call
    return func(*args, **kwargs)
mujoco.FatalError: an OpenGL platform library has not been loaded into this process, this most likely means that a valid OpenGL context has not been created before mjr_makeContext was called`

saran-t, 2022-10-17:
The relevant error message is right at the top of your paste: `GLFWError: (65544) b'X11: The DISPLAY environment variable is missing'`

Most likely culprit is that you're trying to render something over SSH without X forwarding using GLFW. If that's the case, setting `MUJOCO_GL=egl` might help if you have an Nvidia card.

kevinzakka, 2022-10-19:
@hydro-man Just prepend your bash command with `MUJOCO_GL=egl`.

RayYoh, 2022-12-17:
The same error, but useless.

dinghongsong, 2023-03-03:
Boys i got the solutions!
Try to use [X Virtual Framebuffer](https://en.wikipedia.org/wiki/Xvfb#:%7E:text=Xvfb%20or%20X%20virtual%20framebuffer,without%20showing%20any%20screen%20output.) for simulating display environment. 

**Remember to unset the MUJOCO_GL env variable when using xvfb!!!**

For example, run the following commands and it works:
```
xvfb-run -a python train.py --domain_name cartpole --task_name balance --encoder_type pixel --decoder_type pixel --action_repeat 4 --save_video --save_tb --work_dir ./log --seed 1
```

RayYoh, 2023-03-03:
> MUJOCO_GL=egl

Thx for your reply. I add MUJOCO_GL=egl in my .bashrc, and it works.