google-deepmind/dm_control #351: PyMJCF reorders actuators

vaxenburg, 2022-10-25:
MuJoCo and PyMJCF behaviors don't match.

When this XML string is compiled with `mujoco`, an error is thrown, which is the expected behavior:

> "ValueError: Error: stateless actuators must come before stateful actuators"

```python
model = """
<mujoco>
    <worldbody>
      <body name="sphere">
        <joint name="hinge1"/>
        <joint name="hinge2" axis="0 1 0"/>
        <geom name="sphere" size="1"/>
      </body>
    </worldbody>
    <actuator>
      <general name="stateful" joint="hinge1" dyntype="filter" dynprm="1"/>
      <general name="stateless" joint="hinge2" dyntype="none"/>
    </actuator>
</mujoco>
"""

# Compile with MuJoCo.
physics = mujoco.Physics.from_xml_string(model)  # Throws error.
```

However, compilation via `mjcf` works because pymjcf will reorder the actuators:
```python
# Compile with PyMJCF.
mjcf_model = mjcf.from_xml_string(model)
physics = mjcf.Physics.from_mjcf_model(mjcf_model)  # No error thrown here.
```
Calling `mjcf_model.to_xml_string()` will reveal reordered actuators:
```
<mujoco model="unnamed_model">
  <default>
    <default class="/"/>
  </default>
  <worldbody>
    <body name="sphere">
      <joint name="hinge1" class="/"/>
      <joint name="hinge2" class="/" axis="0 1 0"/>
      <geom name="sphere" class="/" size="1"/>
    </body>
  </worldbody>
  <actuator>
    <general name="stateless" class="/" joint="hinge2" dyntype="none"/>
    <general name="stateful" class="/" joint="hinge1" dyntype="filter" dynprm="1"/>
  </actuator>
</mujoco>
```

Moreover, PyMJCF will reorder the actuators even when the "actually_stateful" actuator is made stateful by inheriting `dyntype` from defaults (MuJoCo, on the other hand, will correctly recognize this actuator as stateful and will compile the model correctly and preserve the order of actuators):
```python
model = """
<mujoco>
    <default>
      <default class="state">
        <general dyntype="filter" dynprm="1"/>
      </default>
    </default>
    <worldbody>
      <body name="sphere">
        <joint name="hinge1"/>
        <joint name="hinge2" axis="0 1 0"/>
        <geom name="sphere" size="1"/>
      </body>
    </worldbody>
    <actuator>
      <general name="stateful" joint="hinge1" dyntype="filter" dynprm="1"/>
      <general name="actually_stateful" class="state" joint="hinge2"/>
    </actuator>
</mujoco>
"""

# Compile with PyMJCF.
mjcf_model = mjcf.from_xml_string(model)
physics = mjcf.Physics.from_mjcf_model(mjcf_model)  # Still reorders stateful actuators.
```
Calling `mjcf_model.to_xml_string()`:
```
<mujoco model="unnamed_model">
  <default>
    <default class="/">
      <default class="state">
        <general dyntype="filter" dynprm="1"/>
      </default>
    </default>
  </default>
  <worldbody>
    <body name="sphere">
      <joint name="hinge1" class="/"/>
      <joint name="hinge2" class="/" axis="0 1 0"/>
      <geom name="sphere" class="/" size="1"/>
    </body>
  </worldbody>
  <actuator>
    <general name="actually_stateful" class="state" joint="hinge2"/>
    <general name="stateful" class="/" joint="hinge1" dyntype="filter" dynprm="1"/>
  </actuator>
</mujoco>
```

yuvaltassa, 2022-11-16:
@saran-t, apparently e3d4c0aecfd710cd7428eb77af203dc5fbdacc0f only partially resolved this.

The second MWE still happens, there is still some reordering somewhere...

vaxenburg, 2022-11-17:
@saran-t @yuvaltassa, I'm sorry, this latter problem was on my end actually. There seem to be no reordering anymore and the issue is resolved in e3d4c0aecfd710cd7428eb77af203dc5fbdacc0f. Thanks!

closed 2022-11-17 by yuvaltassa