google-deepmind/dm_control #373: Mujoco, dm_control, EGL Errors

shneorkairois, 2022-12-19:
Hello, whatever the version of dm_control I set I keep getting this errors.

`env: MUJOCO_GL=egl
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[9], line 4
      1 get_ipython().run_line_magic('env', 'MUJOCO_GL=egl')
      2 from typing import Sequence, Tuple
----> 4 from dm_control import suite
      5 from dm_control.rl import control
      6 from IPython.display import HTML

File ~/.local/lib/python3.8/site-packages/dm_control/suite/__init__.py:24
     20 import itertools
     22 from dm_control.rl import control
---> 24 from dm_control.suite import acrobot
     25 from dm_control.suite import ball_in_cup
     26 from dm_control.suite import cartpole

File ~/.local/lib/python3.8/site-packages/dm_control/suite/acrobot.py:20
     16 """Acrobot domain."""
     18 import collections
---> 20 from dm_control import mujoco
     21 from dm_control.rl import control
     22 from dm_control.suite import base

File ~/.local/lib/python3.8/site-packages/dm_control/mujoco/__init__.py:18
      1 # Copyright 2017 The dm_control Authors.
...
---> 87 raw_eglQueryString = _p.PLATFORM.EGL.eglQueryString
     88 raw_eglQueryString.restype = ctypes.c_char_p
     89 raw_eglQueryString.__doc__ = """Raw version of eglQueryString that does not check for availability"""

AttributeError: 'NoneType' object has no attribute 'eglQueryString'`

I am desperate to know if someone has found a fix.

ToasterSC, 2023-03-20:
Excuse me, I also encountered this problem. Could you tell me if you have solved it?

shneorkairois, 2023-03-20:
I haven't, I would however advise you to follow the following steps. My understanding has been that, I had a deep peer dependency issue. I moved to python 3.9, and went from Ubuntu to Debian since, then I haven't encountered the problem.
#136 
> I solved the issue with the following Dockerfile:
> 
> ```
> # Use the official image as a parent image
> FROM tensorflow/tensorflow:2.0.0-gpu-py3
> 
> ENV MJLIB_PATH=/tf/mujoco200_linux/bin/libmujoco200.so
> ENV MJKEY_PATH=/tf/mujoco200_linux/bin/mjkey.txt
> ENV MUJOCO_GL=osmesa
> ENV LD_LIBRARY_PATH=/tf/mujoco200_linux/bin/
> ENV PYOPENGL_PLATFORM=osmesa
> 
> # i need this for ffmpeg
> ENV PATH="/tf/.local/bin/:${PATH}"
> 
> RUN apt-cache search mesa
> RUN apt-get update && apt-get install -y libgl1-mesa-glx libosmesa6
> 
> # Run the command inside your image filesystem
> RUN pip install --upgrade pip && \
>         pip install gym && \
>         pip install gym[atari] && \
>         pip install PyOpenGL==3.1.0 && \
>         pip install dm_control && \
>         pip install tqdm && \
>         pip install np_utils && \
>         pip install opencv-python
> ```

ToasterSC, 2023-03-21:
Thank you very much! I will have a try.

return-sleep, 2023-08-28:
> Thank you very much! I will have a try.

 if you have solved it? I have tried the solution, but haven't solved the problem