google-deepmind/dm_control #382: How do I add geometry to the scene at runtime in mujoco environments?

bobakfb, 2023-02-25:
Hi, I am trying to add some annotations to my scene in form of primitive geometric objects. I have a pretty training/eval loop already that renders out frames, but I'd like to be able to annotate my env and render another frame after I get the normal one, then remove these annotations.

As a basic task, I was looking to add an arrow (for now just a line segment) to maze2d which indicates the agent's velocity.

I saw [this post](https://github.com/deepmind/mujoco/issues/204) previously which shows a tutorial, but following that tutorial doesn't seem to work as `scene.geoms` does not exist.

I looked around a bit and think I almost am there with this snippet: 

```
rgba = np.array((1, 0, 0, 1)) # Red color
line_geom = mujoco.MjvGeom()
line_geom.type = 5 # https://github.com/deepmind/mujoco/blob/main/include/mujoco/mjmodel.h#L87-L109
line_geom.size = [v[0], v[1], 1]
line_geom.pos = [x[0], x[1], .5]
data = sim.data

data.geom_xpos = np.concatenate([data.geom_xpos, [line_geom.pos]])
data.geom_size = np.concatenate([data.geom_size, [line_geom.size]])
data.geom_type = np.concatenate([data.geom_type, [line_geom.type]])
data.geom_rgba = np.concatenate([data.geom_rgba, [rgba]])
```

I am able to create the line geometry, but can't seem to add it to my environment because I can not write to any of the memory controlled by `data`. I looked around but couldn't find a method for appending geometry. Is there anyway to do this? Thanks!

yuvaltassa, 2023-02-25:
Can you explain why you can't draw things in the `mjvScene`, as in the [tutorial](https://github.com/deepmind/mujoco#getting-started)?

bobakfb, 2023-02-25:
Hi @yuvaltassa , I realized my problem was that I have both mujoco and mujoco_py loaded in the same project somehow, and it was confusing me as they are different APIs.

It seems that I needed to follow the directions for mujoco_py. If anyone else finds themselves wanted to do this and the tutorial isn't working, you might be using mujoco_py and the easiest way to do this is with annotations.

closed 2023-02-25 by bobakfb