google-deepmind/dm_control #385: Physical meaning of state and observation in each task

dchen48, 2023-03-09:
From the paper: https://arxiv.org/pdf/2006.12983.pdf, we only know the dimensions of state, observation, and action in each task. Is there a documentation where we can get the physical meaning of state and observations?

yuvaltassa, 2023-03-12:
Read the Python env definition files. For the control suite for example, this is the [cart-pole] definition.(https://github.com/deepmind/dm_control/blob/774f46182140106e22725914aad3c6299ed91edd/dm_control/suite/cartpole.py#L130)

closed 2023-03-12 by yuvaltassa

dchen48, 2023-03-13:
> Read the Python env definition files. For the control suite for example, this is the [cart-pole](https://github.com/deepmind/dm_control/blob/774f46182140106e22725914aad3c6299ed91edd/dm_control/suite/cartpole.py#L130) definition.

Thanks for the quick reply. Can you please help me verify if the observation in cart-pole is [cart position, cos(angle), sin(angle), cart velocity, pole angular velocity]? Also, what are the observations for 2 poles and 3 poles? Thank you very much.