google-deepmind/dm_control #386: Cannot see corridor side planes

RayYoh, 2023-03-13:
When I set `visible_side_planes = True` in `corridor_arenas.EmptyCorridor(visible_side_planes = True)`, I still cannot see the planes when I render the environment. But if I modify
```
self._left_plane = self._mjcf_root.worldbody.add(
        'geom', type='plane', xyaxes=[1, 0, 0, 0, 0, 1], size=[1, 1, 1],
        rgba=[1, 0, 0, alpha], group=_SIDE_WALLS_GEOM_GROUP)
```
to
```
self._left_plane = self._mjcf_root.worldbody.add(
        'geom', type='plane', xyaxes=[1, 0, 0, 0, 0, 1], size=[1, 1, 1],
        rgba=[1, 0, 0, alpha])
```
I can see them. How can I tackle this problem?

RayYoh, 2023-03-13:
By the way, is there any examples for this env used in [Emergence of Locomotion Behaviours in Rich Environments](https://arxiv.org/pdf/1707.02286.pdf)
[image]

yuvaltassa, 2023-03-13:
You need to turn on the visualisation of the geom group `_SIDE_WALLS_GEOM_GROUP`.

Something like 
```py
scene_option = mujoco.wrapper.core.MjvOption()
scene_option.geomgroup[_SIDE_WALLS_GEOM_GROUP] = True
pixels = physics.render(scene_option=scene_option)
```

closed 2023-03-13 by yuvaltassa