google-deepmind/dm_control #390: I installed it without -e option, but I got cannot import name 'constants' error.

Ishihara-Masabumi, 2023-03-24:
I tried to install this repository as follows:

```
git clone https://github.com/deepmind/dm_control.git
cd dm_control/
pip3 install -r requirements.txt 
```

But, after running the command, the following error occurred.

```
python3 ./dm_control/locomotion/soccer/explore.py
ImportError: cannot import name 'constants' from partially initialized module 'dm_control.mujoco.wrapper.mjbindings' (most likely due to a circular import) (/home/dl/DM_Control2/dm_control/dm_control/mujoco/wrapper/mjbindings/__init__.py)
```

Please let me know how to avoid it.

metric-space, 2023-07-28:
Try not running the command from the root directory

raymondchua, 2024-03-21:
I have the same error as @Ishihara-Masabumi as well. Does anyone know how to overcome it?

XiaohanYang2333, 2024-03-22:
I also have this problem and haven't found a solution yet

metric-space, 2024-03-22:
@raymondchua  I did a fresh install just now and changed the bash command I put on here earlier (included pillow install and changed the proper url). This was on arch linux and I have it working. Please try with the edited commands

XiaohanYang2333, 2024-03-22:
来信已收到

metric-space, 2024-03-22:
Something to remember is `pip install .` activates `setup.py` and does a lot more (and including) than just `pip install -r requirements.txt`

raymondchua, 2024-03-22:
Thanks @metric-space. The error shows up when I try to run the following code:

```
from dm_control import suite
env = suite.load('cartpole', 'swingup')
pixels = env.physics.render()
```

raymondchua, 2024-03-22:
Specifically, the import statement is causing the error:

`from dm_control import suite`