google-deepmind/dm_control #395: question regarding segmentation map's values

hnyu, 2023-03-29:
Hi,

I'm trying to use `Camera.render()` to render a segmentation map to check if an object is in a camera's view. I noticed that the segmentation map might contain very few incorrect geom id values on the border of an object. Below is an example (for a 64x64 segmentation map):

[image]
[image]

Basically the first figure is the returned segmentation map, after mapping each geom id to a random color. The view is from a robot camera and the two vertical bars are its two grippers which we can ignore here. If we look at the "basket" on the ground we can see that there are  7 different colors (geom ids). However, the basket is expected to only have 5 geoms (colors). 

The second figure is after I mapped the geom ids to their object indices, and we can see that there are four pixels at the bottom are thought to come from another object (behind the camera). 

This observation occurs even for higher-resolution maps, although in that case the incorrect geom id values are more difficult to be observable. For example, I found that someone else's result at https://github.com/openai/mujoco-py/issues/516 seems to also have this border issue. 

My question is, is this behavior expected?

yuvaltassa, 2023-03-29:
This is due to antialiasing being turned on during segmentation rendering. Of course it *should* be off, and indeed [this line](https://github.com/deepmind/mujoco/blob/2127d0bb3f1ab8c23b9e7799bfa3347ef85f9b62/src/render/render_gl3.c#L833) is supposed to turn it off for segmentation renders, but we have also noticed that this command is sometimes ignored by the GL backend. 

Could you try turning off antialiasing completely in your model by setting [offsamples](https://mujoco.readthedocs.io/en/latest/XMLreference.html#visual-quality-offsamples) to 1? (Or maybe 0, I'm actually not sure)?

hnyu, 2023-03-29:
> This is due to antialiasing being turned on during segmentation rendering. Of course it _should_ be off, and indeed [this [...]

Setting offsamples to 0 indeed solved this problem (I also tried 1 and it seemed to be fine in my case but I'm not sure in general). I wonder if there is any better solution so that it won't affect the RGB frame? That is, we still have anti-aliasing for RGB/depth (buffers that usually have "continuous" values) but not for segmentation.

hnyu, 2023-03-29:
> The correct solution is exactly the line I linked to above. This generally works, yet sometimes doesn't 🤷‍♂️
> 
> What GL backend are you using?

I'm using EGL as the backend.

yuvaltassa, 2023-03-30:
Easiest to just create two `Physics`?

hnyu, 2023-03-30:
> Easiest to just create two `Physics`?

I see. Thanks for the suggestion. I'll just turn off offsamples for now.

closed 2023-03-30 by hnyu