google-deepmind/dm_control #404: UI. A way to zero an actuator in simulate

Melanol, 2023-06-18:
Not very important.

There is a "Clear all" button in simulate to zero all actuators, but I sometimes find it frustrating that there is no way to zero a particular actuator, and I have to try to click as close to 0 as possible, and it is not helpful, as even the smallest value results in movement. This would be useful for manual tests.

[image]

Melanol, 2023-06-18:
Wrong repo. It's for mujoco. Delete it maybe?

yuvaltassa, 2023-06-21:
This is a nice feature request for mjUI. Double-clicking on a slider could reset it. But yes, wrong repo.

closed 2023-06-21 by yuvaltassa