google-deepmind/dm_control #406: Why is friction coefficient not constant?

daichi-3110, 2023-06-26:
Hello, I am a Ph.D student studying MuJoCo.

I simulated sliding an object on the table (the object is moved by mocap body). I calculated normal force and frictional force, and found that the frictional force is different from what I expected. I thought the frictional force is proportional to the normal force until the object stopped, but it was not (attached figure). 

[image]
[image]

I checked the frictional force divided by the normal force, which is maybe friction coefficient, and found the value is not constant. This value should be constant, but not constant in the simulation. Why is friction coefficient not constant??

[image]

The code is attached.
[friction_sample.py.zip](https://github.com/deepmind/dm_control/files/11870261/friction_sample.py.zip)

Thank you very much.

yuvaltassa, 2023-06-27:
That's not how friction works, either in MuJoCo or the real world. The friction force is *in the friction cone*, which means it is *at most* the friction coefficient multiplied by the normal force.

daichi-3110, 2023-06-28:
Thank you for your quick response.

In my knowledge, the static friction is in the friction cone, but the dynamic friction is on the surface of the cone in the real world. Is it right?
In MuJoCo, users can't specify the static friction coefficient and dynamic friction coefficient at the same time. Thus, I think that the dynamic friction is modeled in the way that it is in the friction cone, not on the surface. Am I right??

yuvaltassa, 2023-07-02:
Can you explain how you are forcing the object to move (what do you mean by "the object is moved by mocap body")?

daichi-3110, 2023-07-03:
“Mocap body” is this: https://mujoco.readthedocs.io/en/stable/modeling.html#mocap-bodies
I welded the mocap body to the object, and the object is moved by the following code.

`env.physics.named.data.mocap_pos[0] = [0.1, 0, -0.1]`

daichi-3110, 2023-07-03:
My explanation was not clear, so I will try to explain my question again. I would like to know how to calculate the dynamic friction in MuJoCo. 

To check the dynamic friction, I performed a simulation of an object sliding on a floor. As explained above, the object is controlled by “mocap body” welded to the object. The mocap body was used to slide the object while pressing it against the floor.

The results showed that the friction was not proportional to the normal force. The below figures show the results. The horizontal axis begins at timestep 1. In timestep 0, the object is stopped, and these figures shows the values while the object is moving.

Figure 1 shows the friction while the object is moving, Figure 2 shows the normal force while the object is moving. And Figure 3 shows the dynamic friction coefficient calculated from the friction and normal force. The calculated coefficient varied with time. In my knowledge, the dynamic friction should be constant. Therefore, I could not understand this phenomenon.
[image]

How is the dynamic friction force calculated in MuJoCo?
And, are static friction and dynamic friction calculated as different or same?

yuvaltassa, 2023-07-03:
> And, are static friction and dynamic friction calculated as different or same?

Yes.

Dsai1006, 2023-07-03:
> mocap bodies do not have velocities. By changing the position you are "teleporting" the body.

Thank you for your reply. 

I checked the velocity of the object when moving it using the mocap, and it appears to have velocities.
It takes several iterations to move the object to the position specified by mocap, so it is not "teleporting".
[image]

Source code is here.
[velocity_sample.py.zip](https://github.com/deepmind/dm_control/files/11935434/velocity_sample.py.zip)

Could you please check it?

daichi-3110, 2023-07-03:
> mocap bodies do not have velocities. By changing the position you are "teleporting" the body.
Thank you for your reply.

I checked the velocity of the object when moving it using the mocap, and it appears to have velocities (the below figure).
It takes several iterations to move the object to the position specified by mocap, so it is not "teleporting".

[image]

The source code is here.
[velocity_sample.py.zip](https://github.com/deepmind/dm_control/files/11935473/velocity_sample.py.zip)

Could you please check it?

daichi-3110, 2023-07-03:
> > And, are static friction and dynamic friction calculated as different or same?
> 
> Yes.

"Yes" means that the static friction and dynamic friction are calculated as different, right?

yuvaltassa, 2023-07-04:
Sorry, I misread your question. Dynamic and static friction are the same.

Can you look at the [tutorial colab](https://github.com/deepmind/mujoco#getting-started) please ?

There is a cell there discussing friction with a small example. See if you can recreate the same plots with that. Basically I'm suspecting the business with the weld is not behaving as you think it is.

daichi-3110, 2023-07-04:
Thank you for your quick response.

> Dynamic and static friction are the same.
OK, I understood my result! Thank you.

> Basically I'm suspecting the business with the weld is not behaving as you think it is.
"Mocap body" is teleported itself as you said. But, I think that the body welded with "Mocap body" moves by following this [calculation](https://mujoco.readthedocs.io/en/stable/modeling.html#impedance) as far as I read the link.
How is the motion of the body welded with "Mocap body"  implemented in MuJoCo?

yuvaltassa, 2023-07-24:
You are correct, the welded body's position does get integrated smoothly, but think about the forces it experiences. Every time the mocap body teleports (a little), there is a jump in the force. Basically, mocap bodies are an unphysical thing, so if you compare their behaviour to the real world you are bound to see a mismatch.

closed 2023-07-24 by yuvaltassa