google-deepmind/dm_control #415: Problems with egl rendering

ambitious-octopus, 2023-07-19:
I am working on a mujoco simulation inside a docker container (using nvidia-docker for GPU access). My simulation use egl for hardware rendering and on my local machine works perfectly. I would like to run my simulation on a server with 3 NVIDIA A40. 
When I run this script in the same docker image but on the server
```python
import os
os.environ["MUJOCO_GL"]="egl"
os.environ["MUJOCO_EGL_DEVICE_ID"] = "1"
from dm_control import mujoco
```
I get this error:
```
ImportError: Cannot initialize a headless EGL display.
```

Server:
Nvidia Driver Version: 525.85.12    CUDA Version: 12.0  

Local Machine:
Nvidia Driver Version: 525.116.04   CUDA Version: 12.0  

If I use osmesa (`os.environ["MUJOCO_GL"]="egl"`) everything works correctly. But I would like to use egl. 
I tried to install libnvidia-gl (`apt install libnvidia-gl-525`) as in #214 and #365. but I get the same error. 

I should also add that I do not use NVIDIA Multi-Instance GPU.  
Am I doing something wrong here? 
Thank you in advance for the support!

saran-t, 2023-07-19:
This almost definitely means that your Docker container doesn't have permission to access the GPU hardware specified.

Have you tried setting `MUJOCO_EGL_DEVICE_ID` to 0 ?

ambitious-octopus, 2023-07-19:
I suspected that.  It works with `MUJOCO_EGL_DEVICE_ID=0`. 
I would also add that `nvidia-smi` within the container shows all the gpu's.

xiezhipeng-git, 2024-06-21:
No use doker has this error.
libEGL warning: failed to open /dev/dri/renderD128: Permission denied

libEGL warning: failed to open /dev/dri/card0: Permission denied

 how reslove ?

RealXuChe, 2024-10-18:
I encountered similar issues and error messages when using MuJoCo inside a Docker container on a machine with multiple NVIDIA GPUs. I wrote about the problems I faced and my solutions in a comment under another issue. Hope it helps.

https://github.com/google-deepmind/mujoco/issues/572#issuecomment-2419965230