google-deepmind/dm_control #420: How to get the velocity of each joint in the global coordinate system

dchen48, 2023-08-24:
We can get the (x, y, z) positions in the global coordinate system by physics.data.geom_xpos. How can we get the the velocity of each joint in the global coordinate system? Thank you for the help.

yuvaltassa, 2023-08-24:
Like [this](https://github.com/deepmind/dm_control/blob/d6f9cb4e4a616d1e1d3bd8944bc89541434f1d49/dm_control/mujoco/wrapper/core_test.py#L370).

closed 2023-08-24 by yuvaltassa

dchen48, 2023-08-25:
Thanks a lot for the help. Just to make sure, are the position and the velocity with respect to the global coordinate system (i.e., absolute position and absolute velocity)?

yuvaltassa, 2023-08-28:
Yes.

dchen48, 2023-08-31:
The coordinates returned by physics.data.geom_xpos seems not to follow the order in the corresponding xml file. For example, in the task reacher, there are 10 geoms. The 10th one is the target. However, the 7th one return by physics.data.geom_xpos is the target. What order does physics.data.geom_xpos follow? Thank you.