google-deepmind/dm_control #425: dm-control mujoco tasks use A LOT of cpu's

davidireland3, 2023-09-13:
[Potentially related to this post](https://github.com/google-deepmind/mujoco/issues/203). I was wondering why my code slows down when running a few seeds at a time. When I looked, the CPU's are being throttled by running multiple instances of the MuJoCo environments (specifically I was checking with walker-walk). Using just one environment was maxing out several cores. Is there a way to stop this? I compared with hopper/half-cheetah from the Gymnasium implementations and they use just a single CPU per environment. Do I need to change some setting in dm-control to fix this? 

If any more information is needed (e.g. resource usage from the terminal), then please let me know what else I can add.

davidireland3, 2023-09-14:
closing for now -- seems to be something strange happening where CPU's get maxed only in certain circumstances.

LiuZhenxian123, 2024-03-10:
> closing for now -- seems to be something strange happening where CPU's get maxed only in certain circumstances.

hey david!
how did you solve the problem of high cpu usage? i'm running into the same problem...

LiuZhenxian123, 2024-03-11:
> @LiuZhenxian123 sadly I did not solve it. unless you really need DMC, you could try the OpenAI mujoco tasks. if it is happening to you too, I can reopen.

dm-control seems to try to use all my available CPUs,guess something is wrong with my setting.