google-deepmind/dm_control #428: Call mujoco functions with dm_control.mujoco.wrapper.core.MjModel and dm_control.mujoco.wrapper.core.MjData

tkreiman, 2023-10-08:
Hello,

I would like to call the mujoco.mj_jac function but my model and data are of the type dm_control.mujoco.wrapper.core.MjModel and dm_control.mujoco.wrapper.core.MjData. I have the following code: 

```
body = 'vive_controller'
p_array = np.zeros((3, model.nv), dtype=np.float64)
r_array = np.zeros((3, model.nv), dtype=np.float64)
body_id = model.body_name2id(body)
jac = mujoco.mj_jac(model, data, p_array, r_array, point, body_id)
```

But I get the following error:

`
TypeError: mj_jac(): incompatible function arguments. The following argument types are supported:
    1. (m: mujoco._structs.MjModel, d: mujoco._structs.MjData, jacp: Optional[numpy.ndarray[numpy.float64[m, n], flags.writeable, flags.c_contiguous]], jacr: Optional[numpy.ndarray[numpy.float64[m, n], flags.writeable, flags.c_contiguous]], point: numpy.ndarray[numpy.float64[3, 1]], body: int) -> None

Invoked with: <dm_control.mujoco.wrapper.mjbindings.types.LP_MJMODEL object at 0x7fd62f789940>, <dm_control.mujoco.wrapper.mjbindings.types.LP_MJDATA object at 0x7fd62f785ec0>, array([[0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
        0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.],
       [0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
        0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.],
       [0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
        0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.]]), array([[0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
        0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.],
       [0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
        0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.],
       [0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
        0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.]]), array([-1.95121714e-03,  3.20021586e-01,  2.87537133e+00]), 1`
        
I have also tried passing in model.ptr and data.ptr but I get similar error. How do I recover the original mjmodel and mjdata objects? Thank you.

saran-t, 2023-10-08:
Looks like you're using a rather old version of `dm_control` which isn't compatible with the new (2022) `mujoco` package.

tkreiman, 2023-10-09:
`pip install -U dm_control` did the trick. Simple fix. Thank you.

closed 2023-10-09 by tkreiman