google-deepmind/dm_control #432: (soccer) explore.py gives a stationary frame of the pitch?

Gariscat, 2023-10-23:
Hi, thank you for releasing the code of this fantastic project!

I tried to run the explore.py given in the repo (dm_control/locomotion/soccer/explore.py). However, it seems that the GUI that pops up only renders one stationary frame and nothing is moving. Should I interact with the players using keyboard inputs or load some pretrained policies? What should I do?

I have recorded a 15-second short [video](https://drive.google.com/file/d/1cllPjn2JVxMOKnvF1LdNAxFhh789qT0O/view?usp=share_link) of what happened when I run the explore.py in the command line.

Thanks a lot for any suggestions!!!

yuvaltassa, 2023-10-24:
The physics says "running" but time isn't advancing... hmmm 🤔

What happens when you press the spacebar?

Gariscat, 2023-10-25:
Hi, Yuval!
Thank you for such a quick response. I pressed the spacebar multiple times but still the scene is not changing. However, the "paused" label on the left-buttom corner does swapped from true to false back and forth. What else should I do?

liusiqi43, 2023-10-25:
For context, the players are not supposed to move. However, you should see that the ball falls the ground when the physics engine is stepping. Could you confirm that you don't see that either?

Gariscat, 2023-10-25:
Hi, Siqi!

Yes, I saw the ball falling to the ground. How can I apply some policies to control the agents, no matter pretrained or random?

liusiqi43, 2023-10-25:
I believe you would need to supply a policy instance here that produce
actions for all players that respect the action specs of the environment:
https://github.com/google-deepmind/dm_control/blob/main/dm_control/viewer/__init__.py#L30

Gariscat, 2023-10-25:
OK, I see. I'll give it a try. Thanks again!