google-deepmind/dm_control #449: Segmentation index out of bound

Haichao-Zhang, 2024-02-01:
I used dmc's camera to retrieve the segmentation map of the scene. Sometime, the segmentation index in the seg-map could go out of boundary (> total_num_of_geoms),

```
rgb_buffer = rgb_buffer.astype(np.uint32)
segimage = (image3[:, :, 0] + image3[:, :, 1] * (2**8) + image3[:, :, 2] * (2**16))
```

The issue was sometimes ```segimage > num_of_total_geoms```.

Can you give a hint on this issue?

yuvaltassa, 2024-02-02:
Are you actually using our code for the segmentation image or rolling your own? Geom indices in the seg image do not correspond to modelled geoms. E.g. a site in the model is a geom in the seg image.

Look at how we process the seg image in the Camera class.

yuvaltassa, 2024-02-02:
Read [this block](https://github.com/google-deepmind/dm_control/blob/9e360bf8a069868107e52960a97dcafe2292113f/dm_control/mujoco/engine.py#L902)

closed 2024-02-02 by yuvaltassa

Haichao-Zhang, 2024-02-02:
@yuvaltassa  Thanks for the guidance. I'm using my own code (a class derived from dmc.Camera class). I will double check it.