google-deepmind/dm_control #453: how can I find out information about which body part the observation belongs to?

davidireland3, 2024-03-20:
similar to the OpenAI mujoco tasks where they have documentation detailing which dimension of the state corresponds to which part of the physical body, it would be good to be able to have something similar for DMC. If I look at e.g. walker-walk all I get from the time step is an observation with 'orientations', 'height' and 'velocity'. Can I at least find this information out by looking at the source code?

pvskand, 2024-04-09:
@davidireland3 were you able to figure out how to get the joint sensor level information?

davidireland3, 2024-04-11:
@pvskand kind of. I had to look at the source code when observations were created. I managed for walker but a more complex environment such as quadruped was not as easy, especially when trying to figure out which action related to which actuator.

pvskand, 2024-04-11:
I see. If it's possible could you share the sensor/joint level information and which index does it correspond to in the observation `orientations` and `velocity`? (`height` is a 1-dim quantity so that is trivial).

Thanks in advance!

davidireland3, 2024-04-12:
I have this for walker:
```
    "walker-walk": {
        "body_ids": {
            'torso': np.array([0, 1, 14]),
            'rthigh': np.array([2, 3, 15]),
            'rleg': np.array([4, 5, 16]),
            'rfoot': np.array([6, 7, 17]),
            'lthigh': np.array([8, 9, 18]),
            'lleg': np.array([10, 11, 19]),
            'lfoot': np.array([12, 13, 20]),
        }
```
where the indices correspond to a state vector obtained by doing `np.concatenate([v.flatten() for v in self._time_step.observation.values()])`

I didn't break down whether they correspond to orientations or velocity, I was just interested which aspect of the state corresponds to which body parts.